linuxboot/heads #6: Root filesystem should be read only and verified

osresearch, 2016-08-03:
dm-verity can be used to sign the root filesystem image with the user's key (stored in the ROM).

osresearch, 2016-08-16:
Discussion on qubes mailing list: https://groups.google.com/forum/#!topic/qubes-devel/hG93VcwWtRY

Somewhat working, but needs a way to automate the setup.

tlaurion, 2017-10-03:
Any documentation draft for this for Qubes 3/4?

tlaurion, 2017-10-16:
Documentation is non-existent, scripts are cryptic. Anyone using that feature that could give basic guidelines? Would gladly help.

osresearch, 2017-10-16:
I haven't looked into updating the patches for Qubes 4, nor have I been successful in convincing the Qubes team to make it an option (or better, the default!).

Qubes 3.2 has problems with templates mounted from RO drives if they are re-mounted RW while the template is running.  This made it difficult to use this on a live system if the templates were ever updated while things were in use.

tlaurion, 2017-11-27:
@osresearch Looking forward to replicate dm-verity setup :) If you need help with documentation, just shoot technical stuff, i'm planning on helping documenting everything missing :).

tlaurion, 2018-11-08:
@osresearch : ping :)

tlaurion, 2020-03-09:
@osresearch could be done by merging kernel changes for thin-provisioning support and #676 if some help was given. 

Another interesting path would be to bundle wyng inside of Heads, so the user could actually check his deployment integrity and emergency recover from those backups, which I think is more interesting then simply have dm-verity on TemplateVMs, which would be possible between TemplateVMs upgrades with dm-verity alone.

Thoughts welcome.

tlaurion, 2024-04-16:
Qubesos has no plan on seperaring rootfs from configs, logs or as of now.

Feel free to reopen when upstream OSes (eg Silverblue) static rootfs needs to be verified by Heads, or when qubesos changed their view and are going in a direction where that is possible to implement.

closed 2024-04-16 by tlaurion