linuxboot/heads #7: initrd depends on system libc

osresearch, 2016-08-03:
This makes the builds harder to reproduce and the initrd larger than necessary. uclibc would be a better choice.

osresearch, 2016-11-29:
Also considering using Alpine Linux in a VM -- #54

zaolin, 2016-11-30:
So alpine is too much work. Maybe I can try to setup a build environment for the initramfs only in arch linux. Or do you have other suggestions ?

zaolin, 2016-11-30:
There is also nixos which consists of reproducible binaries only. But it's far away from being stable. Maybe this is a better solution instead of arch linux.

osresearch, 2016-12-28:
After recent checkins (https://github.com/osresearch/heads/commit/5fd9878d28cd05f89551359cc0de1adfb67a310b) the only remaining system dependencies for libraries are `libc.so` and `ld-linux-x86-64.so` (on ubuntu build, fedora build also brings in `libdl.so`).  Might be time to consider building `libmusl-libc`.

closed 2017-01-04 by osresearch