linuxboot/heads #9: Coreboot crossgcc is not verified

osresearch, 2016-08-03:
The Coreboot crossgcc target is not verified and does not check certs.

```
download_showing_percentage() {
        url=$1
        printf " ..${red}  0%%"
        wget --no-check-certificate $url 2>&1 | while read line; do
                printf "${red}"
                echo $line | grep -o "[0-9]\+%" | awk '{printf("\b\b\b\b%4s", $1)}'
                printf "${NC}"
        done
}
```

osresearch, 2016-08-03:
Looks like this is currently pending review: https://review.coreboot.org/#/c/15170/

zaolin, 2016-10-06:
I am on it. The next 4 weeks I have time do that in my holidays ;)

zaolin, 2016-10-22:
Should be done. I pushed some updates. Should get merged in some days..

osresearch, 2016-11-29:
Any update on your patches?  It looks like the current `coreboot/util/crossgcc/buildgcc` still doesn't validate any hashes after downloading: https://github.com/coreboot/coreboot/blob/2c6a8060da994bb22eb1619d55ee74be096682b5/util/crossgcc/buildgcc#L300

zaolin, 2016-11-29:
I am on it. Took longer than expected. Hopefully it gets merged this week.

merge, 2018-05-23:
It seem like [this commit](https://github.com/coreboot/coreboot/commit/92d483a8927066ef16603023be05ad1fb7dea644) implemented this.

closed 2018-05-24 by osresearch