linuxboot/heads #12: SPI flash BP3-0 bits are not set

osresearch, 2016-08-12:
Once the bootblock is more stabilized, the BP3-0 bits should be set to avoid overwriting it and prevent the root of trust from being subverted by a software attack on the flash chip.

zaolin, 2016-10-06:
Works only for the FDT, GBE and BIOS region. For this we need to modify the spi driver for status register support. I have some patches for flashrom itself which can be refactored.

zaolin, 2016-10-22:
As always for flashrom patches published, nothing gets merged 
https://blogs.coreboot.org/blog/author/hatim/
https://www.flashrom.org/pipermail/flashrom/2016-July/014717.html
https://www.flashrom.org/pipermail/flashrom/2016-July/014737.html