linuxboot/heads #16: USB modules should be measured

osresearch, 2016-09-09:
The Linux kernel bootloader should not have USB enabled; loading the modules should require measurement and adjustment of the PCRs (to invalidate the disk encryption keys).

osresearch, 2016-09-27:
Splitting the USB system into a module increased the size of the initrd by 2 MB. That won't work, unless we're much choosier about which modules to include.

jonathanfemideer, 2017-03-01:
At the moment, the installation instructions expect the user to flash Heads to the motherboard, and then to boot heads, mount a USB drive, and install Qubes from that drive.

> The Linux kernel bootloader should not have USB enabled

Perhaps I am misunderstanding, but how would the user install Qubes if Heads does not have USB enabled?

closed 2017-03-31 by osresearch

tlaurion, 2018-03-26:
The point made here is to have usb kernel support as module versus built-in
in the kernel. As a module, it is loaded only if required and requested by
the user (u key at prompt) instead of being there by default when the Linux
payload starts.

I'm pretty sure that issue is fixed now, since the modules to be loaded by
the scripts depends of the configuration of the board and its Linux
configuration.

Either way, USB support is there as we speak as modules. This issue should
be closed. Space constraints have been dealt with since 2016.