linuxboot/heads #17: Firmware updates without reflashing

osresearch, 2016-09-09:
It should be possible to update the firmware (other than the locked boot block) without a hardware programmer.  Perhaps reserve the 4 MB upper flash chip for the bootblock, coreboot and the Linux recovery image, and use the lower 8 MB chip for the normal boot image?

zaolin, 2016-10-06:
Currently a signed update mechanism isn't implemented. You can directly flash via flashrom.org in your operating system. I wrote some code for a signature based update mechanism but this isn't so easy.

osresearch, 2016-11-29:
My new thought is to build a truly minimal recovery image that has a totally stripped Linux kernel, a USB driver and a flashrom like tool (probably my own spiflash writer).  If everything goes pearshaped in the real flash, this fallback system can be booted with the PRR's unlocked and a new image read from an external disk.

The recovery image and the coreboot bootblock would live in a hardware write protected region of the SPI flash, so even with fully unlocked access via the PCH, the CPU would not be able to overwrite the recovery image or circumvent the root of trust.  No signatures are necessary on the firmware flashing since the measurements are what prevent malicious updates from being able to unseal secrets from the TPM.

On something like the Chell this is no problem: 16 MB is plenty of space.  On the x230 with no ME we can recover a few more MB to make this possible.  The top 4 MB chip can be hardware write protected and store the initial boot code/recovery image, while the lower 8 MB chip will store the full featured Heads bootloader.

osresearch, 2017-03-30:
The recovery image doesn't fit into the ROM, but this at least allows users to rewrite their ROMs from the Heads shell.

closed 2017-03-30 by osresearch

osresearch, 2017-04-01:
Started on a draft: http://osresearch.net/Upgrading