linuxboot/heads #26: Disable the ME

osresearch, 2016-09-12:
What happens if the ME firmware is zeroed out? Does coreboot still start up ok?

osresearch, 2016-09-12:
This would free up almost 8 MB for payload, allowing the top 4 MB to be "locked" as the fall back firmware.

Initial tests work, although coreboot does report that "`WARNING: ME has bad firmware`". Discussion on the coreoot list: https://www.coreboot.org/pipermail/coreboot/2016-September/082016.html

osresearch, 2016-09-15:
Maybe we can't disable it, but we can greatly reduce the amount of space it uses: https://www.coreboot.org/pipermail/coreboot/2016-September/082038.html

The flash descriptor can be adjust to reduce the size of the ME region, the BIOS region expanded and many of the problematic ME applications are removed from the ROM.  It's not known what is still in the CPU, but all of the network stack is gone.

osresearch, 2016-09-15:
For future research, the Huffman encoded blobs can be decoded with http://io.smashthestack.org/me/

osresearch, 2016-09-15:
Hacker News discussion: https://news.ycombinator.com/item?id=12489544

zaolin, 2016-11-28:
https://github.com/zaolin/me_cleaner

osresearch, 2016-11-29:
Additional HN discussion: https://news.ycombinator.com/item?id=13056997

Also on hackaday: http://hackaday.com/2016/11/28/neutralizing-intels-management-engine/

The cleaner scripts look good; I should add the extra code for erasing the LZMA modules so that only the huffman ones remain.

closed 2016-11-29 by osresearch