linuxboot/heads #33: Building system fails

tlaurion, 2016-09-24:
A make on a cclean cloned copy of git produce those errors under ubuntu 16.04:
sudo apt-get install m4 bison flex zlib1g-dev git build-essential
git clone https://github.com/osresearch/heads.git

1- initrd/bin is not created:
make
[...]
cmp --quiet "initrd/bin/kexec" "/media/user/7725fbc2-4b2d-4ade-9f1d-337ad8203dc3/Qubes/heads/build/kexec-tools-2.0.12/build/sbin/kexec" || cp -a "/media/user/7725fbc2-4b2d-4ade-9f1d-337ad8203dc3/Qubes/heads/build/kexec-tools-2.0.12/build/sbin/kexec" "initrd/bin/kexec"
cp: cannot create regular file 'initrd/bin/kexec': No such file or directory
Makefile:121: recipe for target 'initrd/bin/kexec' failed
make: *** [initrd/bin/kexec] Error 1

fix:
mkdir -p initrd/bin/

2- initrd/lib/x86_64-linux-gnu is not created:
make
[...]
cmp --quiet "initrd/lib/x86_64-linux-gnu/libtpm.so" "/media/user/7725fbc2-4b2d-4ade-9f1d-337ad8203dc3/Qubes/heads/build/tpmtotp-git/libtpm/libtpm.so" || cp -a "/media/user/7725fbc2-4b2d-4ade-9f1d-337ad8203dc3/Qubes/heads/build/tpmtotp-git/libtpm/libtpm.so" "initrd/lib/x86_64-linux-gnu/libtpm.so"
cp: cannot create regular file 'initrd/lib/x86_64-linux-gnu/libtpm.so': No such file or directory
Makefile:124: recipe for target 'initrd/lib/x86_64-linux-gnu/libtpm.so' failed
make: *** [initrd/lib/x86_64-linux-gnu/libtpm.so] Error 1

fix: mkdir -p initrd/lib/x86_64-linux-gnu/

tlaurion, 2016-09-24:
The following is linked to the linux payload being too big to be added to coreboot. How did you get around it?

Intel ACPI Component Architecture
ASL+ Optimizing Compiler version 20160318-64
Copyright (c) 2000 - 2016 Intel Corporation

coreboot toolchain v1.42 August 11th, 2016
dsdt.aml    720:  Method(XDSS, 2)
Remark   2146 -            ^ Method Argument is never used (Arg1)

dsdt.aml   1250:  Method(POSC,3,Serialized)
Remark   2146 -            ^ Method Argument is never used (Arg0)

dsdt.aml   1696:  Method (_CRS, 0)
Remark   2120 -             ^ Control Method should be made Serialized (due to creation of named objects within)

dsdt.aml   2044: Method(BINF, 2, NotSerialized)
Remark   2120 -           ^ Control Method should be made Serialized (due to creation of named objects within)

ASL Input:     dsdt.aml - 2726 lines, 48015 bytes, 1201 keywords
AML Output:    dsdt.aml - 13847 bytes, 476 named objects, 725 executable opcodes

Compilation complete. 0 Errors, 0 Warnings, 4 Remarks, 394 Optimizations, 2 Constants Folded
    IASL       build/dsdt.aml disassembled correctly.
    HOSTCC     cbfstool/cbfstool.o
    HOSTCC     cbfstool/compress.o
    HOSTCC     cbfstool/cbfs_image.o
    HOSTCC     cbfstool/cbfs-mkstage.o
    HOSTCC     cbfstool/cbfs-mkpayload.o
    HOSTCC     cbfstool/fit.o
    HOSTCC     cbfstool/partitioned_file.o
    HOSTCC     cbfstool/cbfs.o
    HOSTCC     cbfstool/fsp_relocate.o
    HOSTCC     cbfstool/lz4_wrapper.o
    HOSTCC     cbfstool/mem_pool.o
    HOSTCC     cbfstool/region.o
    HOSTCC     cbfstool/lzma.o
    HOSTCC     cbfstool/LzFind.o
    HOSTCC     cbfstool/LzmaDec.o
    HOSTCC     cbfstool/LzmaEnc.o
    HOSTCC     cbfstool/2sha_utility.o
    HOSTCC     cbfstool/2sha1.o
    HOSTCC     cbfstool/2sha256.o
    HOSTCC     cbfstool/2sha512.o
    HOSTCC     cbfstool/lz4.o
    HOSTCC     cbfstool/lz4hc.o
    HOSTCC     cbfstool/lz4frame.o
    HOSTCC     cbfstool/xxhash.o
    HOSTCC     cbfstool/linux_trampoline.o
    HOSTCC     cbfstool/cbfs-payload-linux.o
    HOSTCC     cbfstool/cbfstool (link)
Performing operation on 'COREBOOT' region...
Created CBFS (capacity = 4194008 bytes)
Performing operation on 'COREBOOT' region...
Performing operation on 'COREBOOT' region...
    CBFS       fallback/romstage
Performing operation on 'COREBOOT' region...
E: Input file size (102404) greater than page size (65536).
    CBFS       mrc.cache
Performing operation on 'COREBOOT' region...
    CBFS       cpu_microcode_blob.bin
Performing operation on 'COREBOOT' region...
    CBFS       fallback/ramstage
Performing operation on 'COREBOOT' region...
    CBFS       cmos.default
Performing operation on 'COREBOOT' region...
    CBFS       cmos_layout.bin
Performing operation on 'COREBOOT' region...
    CBFS       fallback/dsdt.aml
Performing operation on 'COREBOOT' region...
    CBFS       fallback/payload
Performing operation on 'COREBOOT' region...
E: The stage file is not in ELF format!
E: Could not add [./bzImage, 4083201 bytes (3987 KB)@0x0]; too big?
E: Failed to add './bzImage' into ROM image.
E: Failed while operating on 'COREBOOT' region!
E: The image will be left unmodified.
Makefile.inc:779: recipe for target 'build/coreboot.pre' failed
make[1]: *** [build/coreboot.pre] Error 1
make[1]: Leaving directory '/media/user/7725fbc2-4b2d-4ade-9f1d-337ad8203dc3/Qubes/heads/build/coreboot-git'
Makefile:87: recipe for target '/media/user/7725fbc2-4b2d-4ade-9f1d-337ad8203dc3/Qubes/heads/build/coreboot-git/build/coreboot.rom' failed

osresearch, 2016-09-26:
I'm doing a clean checkout build now to figure out why the initrd/bin directory isn't created.  The Makefile is supposed to create the directory if it doesn't exist:
```
#
# Install a file into the initrd, if it changed from
# the destination file.
#
define install =
        @if [ ! -d "$(dir $$@)" ]; \
                then mkdir "$(dir $$@)"; \
        fi
        cmp --quiet "$$@" "$$<" || \
        cp -a "$$<" "$$@"
endef
```

osresearch, 2016-09-26:
How large is your Linux kernel and initrd?  The resulting bzImage on ubuntu 15.10 is 3.7MB, so there is only a few hundred KB of spare room in the 4MB image.

In other news, I've managed to reduce the space the ME uses in the second SPI flash chip, which will free up 6MB for the coreboot payload.  One thought is to use the 4 MB one as a write-protected recovery image and store the real kernel in the lower one.

osresearch, 2016-09-26:
On the directory error, `$(dir $$@)` was not expanding correctly.  I've just pushed a fix for it. https://github.com/osresearch/heads/commit/24ef1e1a0ad5f12666f21eeb7c5a128e9c1c9d08

tlaurion, 2016-09-26:
Have a missing libqrencode.so.3 when building linux:
```
[...]
find initrd/bin -type f -a ! -name '*.sh' -print0 \
	| xargs -0 strip
./populate-lib \
	./initrd/lib/x86_64-linux-gnu/ \
	initrd/bin/* \
	initrd/sbin/* \

initrd/bin/qrenc: 'libqrencode.so.3' 	libqrencode.so.3 => not found
[...]
```

osresearch, 2016-09-26:
That is quite a bit larger than mine.
```
diamond:~/build/clean/heads: ls -Fla initrd.cpio 
-rw-rw-r-- 1 hudson hudson 5658120 Sep 26 17:01 initrd.cpio
```

The qrencode looks like a ldd mistake on my part -- it is linked against `../qrencode-3.4.4/.libs/libqrencode.so`, but the initrd building pulls in mine from /usr/lib.

To help track down what files are so large in your initrd, here is what I end up with in a clean checkout:

```
-rw-rw-r-- 1 hudson hudson     110 Sep 26 15:37 ./initrd/.ash_history
-rwxrwxr-x 1 hudson hudson   10376 Sep 26 17:01 ./initrd/bin/base32*
-rwxr-xr-x 1 hudson hudson  306992 Sep 26 17:01 ./initrd/bin/busybox*
-rwxrwxr-x 1 hudson hudson   23240 Sep 26 17:01 ./initrd/bin/cbmem*
-rwxrwxr-x 1 hudson hudson   10400 Sep 26 17:01 ./initrd/bin/clearown*
-rwxrwxr-x 1 hudson hudson   10432 Sep 26 17:01 ./initrd/bin/counter_create*
-rwxrwxr-x 1 hudson hudson   10416 Sep 26 17:01 ./initrd/bin/counter_increment*
-rwxrwxr-x 1 hudson hudson   10392 Sep 26 17:01 ./initrd/bin/counter_read*
-rwxrwxr-x 1 hudson hudson   10432 Sep 26 17:01 ./initrd/bin/counter_release*
-rwxr-xr-x 1 hudson hudson   78752 Sep 26 17:01 ./initrd/bin/dmsetup*
-rwxrwxr-x 1 hudson hudson    6280 Sep 26 17:01 ./initrd/bin/forceclear*
-rwxrwxr-x 1 hudson hudson   22832 Sep 26 17:01 ./initrd/bin/getcapability*
-rwxr-xr-x 1 hudson hudson  372400 Sep 26 17:01 ./initrd/bin/gpgv*
-rwxrwxr-x 1 hudson hudson  167144 Sep 26 17:01 ./initrd/bin/kexec*
-rwxrwxr-x 1 hudson hudson   10464 Sep 26 17:01 ./initrd/bin/nv*
-rwxrwxr-x 1 hudson hudson   14544 Sep 26 17:01 ./initrd/bin/nv_definespace*
-rwxrwxr-x 1 hudson hudson   14584 Sep 26 17:01 ./initrd/bin/nv_readvalue*
-rwxrwxr-x 1 hudson hudson   14560 Sep 26 17:01 ./initrd/bin/nv_writevalue*
-rwxrwxr-x 1 hudson hudson    6296 Sep 26 17:01 ./initrd/bin/pcrreset*
-rwxrwxr-x 1 hudson hudson    6280 Sep 26 17:01 ./initrd/bin/physicaldisable*
-rwxrwxr-x 1 hudson hudson    6280 Sep 26 17:01 ./initrd/bin/physicalenable*
-rwxrwxr-x 1 hudson hudson   14496 Sep 26 17:01 ./initrd/bin/physicalpresence*
-rwxrwxr-x 1 hudson hudson    6272 Sep 26 17:01 ./initrd/bin/physicalsetdeactivated*
-rwxrwxr-x 1 hudson hudson    6264 Sep 26 17:01 ./initrd/bin/qrenc*
-rwxrwxr-x 1 hudson hudson   18688 Sep 26 17:01 ./initrd/bin/sealfile2*
-rwxrwxr-x 1 hudson hudson    1238 Sep 26 15:24 ./initrd/bin/sealtotp.sh*
-rwxrwxr-x 1 hudson hudson   14544 Sep 26 17:01 ./initrd/bin/takeown*
-rwxrwxr-x 1 hudson hudson   10408 Sep 26 17:01 ./initrd/bin/totp*
-rwxrwxr-x 1 hudson hudson   10432 Sep 26 17:01 ./initrd/bin/unsealfile*
-rwxrwxr-x 1 hudson hudson     426 Sep 26 15:24 ./initrd/bin/unsealtotp.sh*
-rwxrwxr-x 1 hudson hudson    1095 Sep 26 15:37 ./initrd/init*
-rwxrwxr-x 1 hudson hudson  154376 Sep 26 17:01 ./initrd/lib/x86_64-linux-gnu/ld-linux-x86-64.so.2*
-rwxrwxr-x 1 hudson hudson   66800 Sep 26 17:01 ./initrd/lib/x86_64-linux-gnu/libbz2.so.1.0*
-rwxrwxr-x 1 hudson hudson 1869392 Sep 26 17:01 ./initrd/lib/x86_64-linux-gnu/libc.so.6*
-rwxrwxr-x 1 hudson hudson  294496 Sep 26 17:01 ./initrd/lib/x86_64-linux-gnu/libdevmapper.so.1.02.1*
-rwxrwxr-x 1 hudson hudson   14592 Sep 26 17:01 ./initrd/lib/x86_64-linux-gnu/libdl.so.2*
-rwxrwxr-x 1 hudson hudson  137400 Sep 26 17:01 ./initrd/lib/x86_64-linux-gnu/liblzma.so.5*
-rwxrwxr-x 1 hudson hudson  378336 Sep 26 15:25 ./initrd/lib/x86_64-linux-gnu/libmbedcrypto.so.0*
-rwxrwxr-x 1 hudson hudson  448440 Sep 26 17:01 ./initrd/lib/x86_64-linux-gnu/libpcre.so.3*
-rwxrwxr-x 1 hudson hudson  142080 Sep 26 17:01 ./initrd/lib/x86_64-linux-gnu/libpthread.so.0*
-rwxrwxr-x 1 hudson hudson   47240 Sep 26 17:01 ./initrd/lib/x86_64-linux-gnu/libqrencode.so.3*
-rwxrwxr-x 1 hudson hudson  101232 Sep 26 17:01 ./initrd/lib/x86_64-linux-gnu/libresolv.so.2*
-rwxrwxr-x 1 hudson hudson   31680 Sep 26 17:01 ./initrd/lib/x86_64-linux-gnu/librt.so.1*
-rwxrwxr-x 1 hudson hudson  138400 Sep 26 17:01 ./initrd/lib/x86_64-linux-gnu/libselinux.so.1*
-rwxrwxr-x 1 hudson hudson  368672 Sep 26 15:25 ./initrd/lib/x86_64-linux-gnu/libtpm.so*
-rwxrwxr-x 1 hudson hudson  122624 Sep 26 17:01 ./initrd/lib/x86_64-linux-gnu/libudev.so.1*
-rwxrwxr-x 1 hudson hudson  104824 Sep 26 17:01 ./initrd/lib/x86_64-linux-gnu/libz.so.1*
-rwxrwxr-x 1 hudson hudson    1373 Sep 26 15:37 ./initrd/mount-boot*
-rwxrwxr-x 1 hudson hudson     519 Sep 26 15:37 ./initrd/start-xen*
-rw-rw-r-- 1 hudson hudson   17559 Sep 26 15:37 ./initrd/trustedkeys.gpg
```

tlaurion, 2016-09-26:
du -sh ./initrd/*/*|sort -h```

4.0K    ./initrd/bin/sealtotp.sh
4.0K    ./initrd/bin/unsealtotp.sh
8.0K    ./initrd/bin/forceclear
8.0K    ./initrd/bin/pcrreset
8.0K    ./initrd/bin/physicaldisable
8.0K    ./initrd/bin/physicalenable
8.0K    ./initrd/bin/physicalsetdeactivated
8.0K    ./initrd/bin/qrenc
12K     ./initrd/bin/base32
12K     ./initrd/bin/clearown
12K     ./initrd/bin/counter_create
12K     ./initrd/bin/counter_increment
12K     ./initrd/bin/counter_read
12K     ./initrd/bin/counter_release
12K     ./initrd/bin/nv
12K     ./initrd/bin/totp
12K     ./initrd/bin/unsealfile
16K     ./initrd/bin/nv_definespace
16K     ./initrd/bin/nv_readvalue
16K     ./initrd/bin/nv_writevalue
16K     ./initrd/bin/physicalpresence
16K     ./initrd/bin/takeown
20K     ./initrd/bin/sealfile2
24K     ./initrd/bin/cbmem
24K     ./initrd/bin/getcapability
148K    ./initrd/bin/dmsetup
160K    ./initrd/bin/kexec
300K    ./initrd/bin/busybox
364K    ./initrd/bin/gpgv
5.2M    ./initrd/lib/x86_64-linux-gnu

```

tlaurion, 2016-09-26:
That's the best I was able to get to your output with du.

du -ah ./initrd | grep -v "/$" | sort -h

```
4.0K    ./initrd/bin/sealtotp.sh
4.0K    ./initrd/bin/unsealtotp.sh
4.0K    ./initrd/init
4.0K    ./initrd/lib64
4.0K    ./initrd/mount-boot
4.0K    ./initrd/sbin
4.0K    ./initrd/start-xen
8.0K    ./initrd/bin/forceclear
8.0K    ./initrd/bin/pcrreset
8.0K    ./initrd/bin/physicaldisable
8.0K    ./initrd/bin/physicalenable
8.0K    ./initrd/bin/physicalsetdeactivated
8.0K    ./initrd/bin/qrenc
12K     ./initrd/bin/base32
12K     ./initrd/bin/clearown
12K     ./initrd/bin/counter_create
12K     ./initrd/bin/counter_increment
12K     ./initrd/bin/counter_read
12K     ./initrd/bin/counter_release
12K     ./initrd/bin/nv
12K     ./initrd/bin/totp
12K     ./initrd/bin/nv                                                                                                         
12K     ./initrd/bin/totp
12K     ./initrd/bin/unsealfile
16K     ./initrd/bin/nv_definespace
16K     ./initrd/bin/nv_readvalue
16K     ./initrd/bin/nv_writevalue
16K     ./initrd/bin/physicalpresence
16K     ./initrd/bin/takeown
16K     ./initrd/lib/x86_64-linux-gnu/libdl.so.2
20K     ./initrd/bin/sealfile2
20K     ./initrd/trustedkeys.gpg
24K     ./initrd/bin/cbmem
24K     ./initrd/bin/getcapability
32K     ./initrd/lib/x86_64-linux-gnu/librt.so.1
68K     ./initrd/lib/x86_64-linux-gnu/libbz2.so.1.0
100K    ./initrd/lib/x86_64-linux-gnu/libresolv.so.2
104K    ./initrd/lib/x86_64-linux-gnu/libz.so.1
124K    ./initrd/lib/x86_64-linux-gnu/libudev.so.1
128K    ./initrd/lib/x86_64-linux-gnu/libselinux.so.1
136K    ./initrd/lib/x86_64-linux-gnu/libpthread.so.0
148K    ./initrd/bin/dmsetup
160K    ./initrd/bin/kexec
160K    ./initrd/lib/x86_64-linux-gnu/ld-linux-x86-64.so.2
300K    ./initrd/bin/busybox
356K    ./initrd/lib/x86_64-linux-gnu/libdevmapper.so.1.02.1
364K    ./initrd/bin/gpgv
364K    ./initrd/lib/x86_64-linux-gnu/libtpm.so
372K    ./initrd/lib/x86_64-linux-gnu/libmbedcrypto.so.0
448K    ./initrd/lib/x86_64-linux-gnu/libpcre.so.3
1.1M    ./initrd/lib/x86_64-linux-gnu/libm.so.6
1.3M    ./initrd/bin
1.8M    ./initrd/lib/x86_64-linux-gnu/libc.so.6
5.2M    ./initrd/lib
5.2M    ./initrd/lib/x86_64-linux-gnu
6.5M    ./initrd
```

tlaurion, 2016-09-26:
I seem to have the file /initrd/lib/x86_64-linux-gnu/libm.so.6 that you don't have.

osresearch, 2016-09-26:
I used `find ./initrd -type f -print0 | xargs -0 ls -Fla` to generate my list.

Can you tell which binary is depending on libm? We really don't have space for another 1MB library...  `ldd initrd/bin/*` should tell you which one brought it in.

osresearch, 2016-09-27:
I built on fedora-23 (qubes 3.2) and found that `dmsetup` is linked against `libm.so.6` and that is making the initrd too large. Commenting it out in the makefile shrank initrd+bzImage from 4.5MB to 4.1MB, but it is still too large.

```
--- a/Makefile
+++ b/Makefile
@@ -151,7 +151,7 @@ $(build)/$(coreboot_dir)/util/cbmem/cbmem: $(build)/$(coreboot_dir)/.canary
 # Mounting dm-verity file systems requires dm-verity to be installed
 # We use gpgv to verify the signature on the root h ash.
 # Both of these should be brought in as modules instead of from /sbin
-initrd_bins += initrd/bin/dmsetup
+#initrd_bins += initrd/bin/dmsetup
 initrd/bin/dmsetup: /sbin/dmsetup
        cp "$<" "$@"
 initrd_bins += initrd/bin/gpgv
```

Other extra libraries look like libcap, libdw, libelf.

tlaurion, 2016-09-27:
Bingo.

osresearch, 2016-09-27:
Since the dm-verity setup (issue #6) isn't ready, you should be able to build without `dmsetup` and have an image to test with.  Good luck!

closed 2016-09-27 by osresearch