linuxboot/heads #36: Flashed heads doesn't pop a shell but searches for a bootable HD partition and panics

tlaurion, 2016-10-03:
[image]

tlaurion, 2016-10-03:
Any draft of that [page](https://trmm.net/Emulating_Heads)?  I've looked [here](https://www.coreboot.org/QEMU_Build_Tutorial) without any debugging satisfaction.

osresearch, 2016-10-04:
[image]

That screenshot looks very familiar... Are you building from the git branch or a release tag? Is the initrd linked into the bzImage or with the coreboot cbfstool?

Can you verify the contents of the `initrd.cpio` file?  I've found that if `/dev` and `/dev/console` is not included then the kernel panic occurs.

```
diamond:~/build/clean/heads: cpio -tv < initrd.cpio  | grep -v ' -> '
drwxrwxr-x   0 root     root            0 Dec 31  1969 .
-rw-rw-r--   0 root     root          110 Dec 31  1969 .ash_history
drwxr-xr-x   0 root     root            0 Dec 31  1969 /dev
crw-------   0 root     root       5,   1 Dec 31  1969 /dev/console
drwxrwxr-x   0 root     root            0 Dec 31  1969 bin
-rwxrwxr-x   0 root     root        10376 Dec 31  1969 bin/base32
-rwxr-xr-x   0 root     root       306992 Dec 31  1969 bin/busybox
-rwxrwxr-x   0 root     root        23240 Dec 31  1969 bin/cbmem
-rwxrwxr-x   0 root     root        10400 Dec 31  1969 bin/clearown
-rwxrwxr-x   0 root     root        10432 Dec 31  1969 bin/counter_create
-rwxrwxr-x   0 root     root        10416 Dec 31  1969 bin/counter_increment
-rwxrwxr-x   0 root     root        10392 Dec 31  1969 bin/counter_read
-rwxrwxr-x   0 root     root        10432 Dec 31  1969 bin/counter_release
-rwxr-xr-x   0 root     root        78752 Dec 31  1969 bin/dmsetup
-rwxrwxr-x   0 root     root         6280 Dec 31  1969 bin/forceclear
-rwxrwxr-x   0 root     root        22832 Dec 31  1969 bin/getcapability
-rwxr-xr-x   0 root     root       372400 Dec 31  1969 bin/gpgv
-rwxrwxr-x   0 root     root       167144 Dec 31  1969 bin/kexec
-rwxrwxr-x   0 root     root        10464 Dec 31  1969 bin/nv
-rwxrwxr-x   0 root     root        14544 Dec 31  1969 bin/nv_definespace
-rwxrwxr-x   0 root     root        14584 Dec 31  1969 bin/nv_readvalue
-rwxrwxr-x   0 root     root        14560 Dec 31  1969 bin/nv_writevalue
-rwxrwxr-x   0 root     root         6296 Dec 31  1969 bin/pcrreset
-rwxrwxr-x   0 root     root         6280 Dec 31  1969 bin/physicaldisable
-rwxrwxr-x   0 root     root         6280 Dec 31  1969 bin/physicalenable
-rwxrwxr-x   0 root     root        14496 Dec 31  1969 bin/physicalpresence
-rwxrwxr-x   0 root     root         6272 Dec 31  1969 bin/physicalsetdeactivated
-rwxrwxr-x   0 root     root         6264 Dec 31  1969 bin/qrenc
-rwxrwxr-x   0 root     root        18688 Dec 31  1969 bin/sealfile2
-rwxrwxr-x   0 root     root         1238 Dec 31  1969 bin/sealtotp.sh
-rwxrwxr-x   0 root     root        14544 Dec 31  1969 bin/takeown
-rwxrwxr-x   0 root     root        10408 Dec 31  1969 bin/totp
-rwxrwxr-x   0 root     root        10432 Dec 31  1969 bin/unsealfile
-rwxrwxr-x   0 root     root          426 Dec 31  1969 bin/unsealtotp.sh
11052 blocks
-rwxrwxr-x   0 root     root         1095 Dec 31  1969 init
drwxrwxr-x   0 root     root            0 Dec 31  1969 lib
drwxrwxr-x   0 root     root            0 Dec 31  1969 lib/x86_64-linux-gnu
-rwxrwxr-x   0 root     root       154376 Dec 31  1969 lib/x86_64-linux-gnu/ld-linux-x86-64.so.2
-rwxrwxr-x   0 root     root        66800 Dec 31  1969 lib/x86_64-linux-gnu/libbz2.so.1.0
-rwxrwxr-x   0 root     root      1869392 Dec 31  1969 lib/x86_64-linux-gnu/libc.so.6
-rwxrwxr-x   0 root     root       294496 Dec 31  1969 lib/x86_64-linux-gnu/libdevmapper.so.1.02.1
-rwxrwxr-x   0 root     root        14592 Dec 31  1969 lib/x86_64-linux-gnu/libdl.so.2
-rwxrwxr-x   0 root     root       137400 Dec 31  1969 lib/x86_64-linux-gnu/liblzma.so.5
-rwxrwxr-x   0 root     root       378336 Dec 31  1969 lib/x86_64-linux-gnu/libmbedcrypto.so.0
-rwxrwxr-x   0 root     root       448440 Dec 31  1969 lib/x86_64-linux-gnu/libpcre.so.3
-rwxrwxr-x   0 root     root       142080 Dec 31  1969 lib/x86_64-linux-gnu/libpthread.so.0
-rwxrwxr-x   0 root     root        47240 Dec 31  1969 lib/x86_64-linux-gnu/libqrencode.so.3
-rwxrwxr-x   0 root     root       101232 Dec 31  1969 lib/x86_64-linux-gnu/libresolv.so.2
-rwxrwxr-x   0 root     root        31680 Dec 31  1969 lib/x86_64-linux-gnu/librt.so.1
-rwxrwxr-x   0 root     root       138400 Dec 31  1969 lib/x86_64-linux-gnu/libselinux.so.1
-rwxrwxr-x   0 root     root       368672 Dec 31  1969 lib/x86_64-linux-gnu/libtpm.so
-rwxrwxr-x   0 root     root       122624 Dec 31  1969 lib/x86_64-linux-gnu/libudev.so.1
-rwxrwxr-x   0 root     root       104824 Dec 31  1969 lib/x86_64-linux-gnu/libz.so.1
drwxrwxr-x   0 root     root            0 Dec 31  1969 lib64
-rwxrwxr-x   0 root     root         1373 Dec 31  1969 mount-boot
drwxrwxr-x   0 root     root            0 Dec 31  1969 sbin
-rwxrwxr-x   0 root     root          519 Dec 31  1969 start-xen
-rw-rw-r--   0 root     root        17559 Dec 31  1969 trustedkeys.gpg
```

tlaurion, 2016-10-04:
Console is included.

```
cpio -tv < initrd.cpio  | grep -v ' -> '
12797 blocks
drwxrwxr-x   0 root     root            0 Dec 31  1969 .
-rw-rw-r--   0 root     root          110 Dec 31  1969 .ash_history
drwxr-xr-x   0 root     root            0 Dec 31  1969 /dev
crw-------   0 root     root       5,   1 Dec 31  1969 /dev/console
drwxrwxr-x   0 root     root            0 Dec 31  1969 bin
-rwxrwxr-x   0 root     root        10448 Dec 31  1969 bin/base32
-rwxr-xr-x   0 root     root       307040 Dec 31  1969 bin/busybox
-rwxrwxr-x   0 root     root        23312 Dec 31  1969 bin/cbmem
-rwxrwxr-x   0 root     root        10464 Dec 31  1969 bin/clearown
-rwxrwxr-x   0 root     root        10496 Dec 31  1969 bin/counter_create
-rwxrwxr-x   0 root     root        10480 Dec 31  1969 bin/counter_increment
-rwxrwxr-x   0 root     root        10456 Dec 31  1969 bin/counter_read
-rwxrwxr-x   0 root     root        10496 Dec 31  1969 bin/counter_release
-rwxrwxr-x   0 root     root         6344 Dec 31  1969 bin/forceclear
-rwxrwxr-x   0 root     root        22896 Dec 31  1969 bin/getcapability
-rwxr-xr-x   0 root     root       372488 Dec 31  1969 bin/gpgv
-rwxrwxr-x   0 root     root       163088 Dec 31  1969 bin/kexec
-rwxrwxr-x   0 root     root        10528 Dec 31  1969 bin/nv
-rwxrwxr-x   0 root     root        14608 Dec 31  1969 bin/nv_definespace
-rwxrwxr-x   0 root     root        14648 Dec 31  1969 bin/nv_readvalue
-rwxrwxr-x   0 root     root        14624 Dec 31  1969 bin/nv_writevalue
-rwxrwxr-x   0 root     root         6360 Dec 31  1969 bin/pcrreset
-rwxrwxr-x   0 root     root         6344 Dec 31  1969 bin/physicaldisable
-rwxrwxr-x   0 root     root         6344 Dec 31  1969 bin/physicalenable
-rwxrwxr-x   0 root     root        14560 Dec 31  1969 bin/physicalpresence
-rwxrwxr-x   0 root     root         6336 Dec 31  1969 bin/physicalsetdeactivated
-rwxrwxr-x   0 root     root         6328 Dec 31  1969 bin/qrenc
-rwxrwxr-x   0 root     root        18752 Dec 31  1969 bin/sealfile2
-rwxrwxr-x   0 root     root         1238 Dec 31  1969 bin/sealtotp.sh
-rwxrwxr-x   0 root     root        14608 Dec 31  1969 bin/takeown
-rwxrwxr-x   0 root     root        10472 Dec 31  1969 bin/totp
-rwxrwxr-x   0 root     root        10496 Dec 31  1969 bin/unsealfile
-rwxrwxr-x   0 root     root          426 Dec 31  1969 bin/unsealtotp.sh
-rwxrwxr-x   0 root     root         1095 Dec 31  1969 init
drwxrwxr-x   0 root     root            0 Dec 31  1969 lib
drwxrwxr-x   0 root     root            0 Dec 31  1969 lib/x86_64-linux-gnu
-rwxrwxr-x   0 root     root       162632 Dec 31  1969 lib/x86_64-linux-gnu/ld-linux-x86-64.so.2
-rwxrwxr-x   0 root     root        66800 Dec 31  1969 lib/x86_64-linux-gnu/libbz2.so.1.0
-rwxrwxr-x   0 root     root      1864888 Dec 31  1969 lib/x86_64-linux-gnu/libc.so.6
-rwxrwxr-x   0 root     root       360664 Dec 31  1969 lib/x86_64-linux-gnu/libdevmapper.so.1.02.1
-rwxrwxr-x   0 root     root        14608 Dec 31  1969 lib/x86_64-linux-gnu/libdl.so.2
-rwxrwxr-x   0 root     root      1088952 Dec 31  1969 lib/x86_64-linux-gnu/libm.so.6
-rwxrwxr-x   0 root     root       378360 Dec 31  1969 lib/x86_64-linux-gnu/libmbedcrypto.so.0
-rwxrwxr-x   0 root     root       456632 Dec 31  1969 lib/x86_64-linux-gnu/libpcre.so.3
-rwxrwxr-x   0 root     root       138744 Dec 31  1969 lib/x86_64-linux-gnu/libpthread.so.0
-rwxrwxr-x   0 root     root       101200 Dec 31  1969 lib/x86_64-linux-gnu/libresolv.so.2
-rwxrwxr-x   0 root     root        31712 Dec 31  1969 lib/x86_64-linux-gnu/librt.so.1
-rwxrwxr-x   0 root     root       130224 Dec 31  1969 lib/x86_64-linux-gnu/libselinux.so.1
-rwxrwxr-x   0 root     root       368712 Dec 31  1969 lib/x86_64-linux-gnu/libtpm.so
-rwxrwxr-x   0 root     root       126840 Dec 31  1969 lib/x86_64-linux-gnu/libudev.so.1
-rwxrwxr-x   0 root     root       104824 Dec 31  1969 lib/x86_64-linux-gnu/libz.so.1
drwxrwxr-x   0 root     root            0 Dec 31  1969 lib64
-rwxrwxr-x   0 root     root         1373 Dec 31  1969 mount-boot
drwxrwxr-x   0 root     root            0 Dec 31  1969 sbin
-rwxrwxr-x   0 root     root          519 Dec 31  1969 start-xen
-rw-rw-r--   0 root     root        17559 Dec 31  1969 trustedkeys.gpg

```

osresearch, 2016-10-04:
Can you `chroot` into it on your build machine?

```
diamond:~/build/clean/heads: sudo chroot ./initrd /bin/ash
[sudo] password for hudson: 
/ # ls
bin              lib              mount-boot       start-xen
init             lib64            sbin             trustedkeys.gpg
/ # 
```

What does cbfstool report when it builds it? The payload section should be about 3.8 MB:

```
Name                           Offset     Type         Size
cbfs master header             0x0        cbfs header  32
cpu_microcode_blob.bin         0x80       microcode    22528
cmos.default                   0x5900     cmos_default 256
cmos_layout.bin                0x5a40     cmos_layout  1952
fallback/dsdt.aml              0x6240     raw          13847
(empty)                        0x98c0     null         26264
fallback/romstage              0xff80     stage        77956
(empty)                        0x23080    null         52760
mrc.cache                      0x2fec0    mrc_cache    65536
fallback/ramstage              0x3ff00    stage        88072
fallback/payload               0x55740    payload      3768865
(empty)                        0x3ed9c0   null         72984
bootblock                      0x3ff700   bootblock    1952

Built lenovo/x230 (ThinkPad X230)
```

tlaurion, 2016-10-04:
sudo chroot ./initrd /bin/ash
/ # ls
bin              init             lib              lib64            mount-boot       sbin             start-xen        trustedkeys.gpg
Coreboot output: 
```
make[1]: Entering directory '/media/user/7725fbc2-4b2d-4ade-9f1d-337ad8203dc3/Qubes/heads/build/coreboot-git'
yes
    GEN        generated/bootblock.ld
    CP         bootblock/arch/x86/bootblock.ld
    LINK       cbfs/fallback/bootblock.debug
    OBJCOPY    cbfs/fallback/bootblock.elf
    OBJCOPY    bootblock.raw.bin
    CREATE     build/mainboard/lenovo/x230/cbfs-file.QBjYJB.out (from src/mainboard/lenovo/x230/cmos.default)
Performing operation on 'COREBOOT' region...
Created CBFS (capacity = 4194008 bytes)
Performing operation on 'COREBOOT' region...
Performing operation on 'COREBOOT' region...
    CBFS       fallback/romstage
Performing operation on 'COREBOOT' region...
E: Input file size (102404) greater than page size (65536).
    CBFS       mrc.cache
Performing operation on 'COREBOOT' region...
    CBFS       cpu_microcode_blob.bin
Performing operation on 'COREBOOT' region...
    CBFS       fallback/ramstage
Performing operation on 'COREBOOT' region...
    CBFS       cmos.default
Performing operation on 'COREBOOT' region...
    CBFS       cmos_layout.bin
Performing operation on 'COREBOOT' region...
    CBFS       fallback/dsdt.aml
Performing operation on 'COREBOOT' region...
    CBFS       fallback/payload
Performing operation on 'COREBOOT' region...
E: The stage file is not in ELF format!

** WARNING **
Coreboot will be built with a fake Intel Firmware Descriptor (IFD).
Never write a complete coreboot.rom with a fake IFD to your board's
flash ROM! Make sure that you only write valid flash regions.

    IFDFAKE    Building a fake Intel Firmware Descriptor
    DD         Adding Intel Firmware Descriptor
    CBFS       coreboot.rom
    CBFSPRINT  coreboot.rom

Performing operation on 'COREBOOT' region...
Name                           Offset     Type         Size
cbfs master header             0x0        cbfs header  32
cpu_microcode_blob.bin         0x80       microcode    22528
cmos.default                   0x5900     cmos_default 256
cmos_layout.bin                0x5a40     cmos_layout  1952
fallback/dsdt.aml              0x6240     raw          13847
(empty)                        0x98c0     null         26264
fallback/romstage              0xff80     stage        77956
(empty)                        0x23080    null         52760
mrc.cache                      0x2fec0    mrc_cache    65536
fallback/ramstage              0x3ff00    stage        88087
fallback/payload               0x55780    payload      3688369
(empty)                        0x3d9f80   null         153432
bootblock                      0x3ff700   bootblock    1952

Built lenovo/x230 (ThinkPad X230)
make[1]: Leaving directory '/media/user/7725fbc2-4b2d-4ade-9f1d-337ad8203dc3/Qubes/heads/build/coreboot-git'
dd if="/media/user/7725fbc2-4b2d-4ade-9f1d-337ad8203dc3/Qubes/heads/build/coreboot-git/build/coreboot.rom" of="x230.rom" bs=1M skip=8
4+0 records in
4+0 records out
4194304 bytes (4.2 MB, 4.0 MiB) copied, 0.0248528 s, 169 MB/s
```

tlaurion, 2016-10-05:
Working. Seems like the past rom was in a weird state. The one and detailed here works! So if I understand [well](https://github.com/QubesOS/qubes-issues/issues/1594#issuecomment-239585006) 

Qubes needs to be installed first prior to even flashing.

osresearch, 2016-10-05:
I didn't commit that patch since `dmsetup` will be necessary for crypt filesystems.  That was a local patch that you could make to let it build on your system (until we build our own dmsetup so that we can control the libraries it links in).

It is possible to install Qubes from the heads shell, but it is a little tricky since you need the modified `xen-4.6.3.gz` executable accessible.  The way I've done it was by adding another partition to my Qubes USB stick.  I tried to document it here: https://trmm.net/Installing_Heads#Installing_Qubes

osresearch, 2016-11-29:
Following up on this -- were you able to get qubes up and running? That falls under issue #27, "Installing Qubes on Heads is hard".

closed 2016-11-29 by osresearch