linuxboot/heads #37: Coreboot config should support multiple targets

osresearch, 2016-10-04:
The makefile target for coreboot config only supports the x230; it should support building the qemu target or the (soon to be create) chell chromebook target without having to copy files around.

zaolin, 2016-10-06:
There is the abuild tool under util/abuild. It can build mutilple devices directly also we are planing to bring up a service for building coreboot images (GPG signed/encrypted) -> https://github.com/zaolin/coreboot-builder

osresearch, 2016-11-28:
Supported as of https://github.com/osresearch/heads/commit/4fbd6ca58bff576c5ba149de472a18615e82a665

Run `make TARGET=qemu` or `make TARGET=x230`  to generate the different ROM images.  Chell support still needs to be merged (minimal changes, but needs to be back tested)

closed 2016-11-28 by osresearch