linuxboot/heads #38: Chell build requires FSP

osresearch, 2016-10-05:
Available from https://downloadmirror.intel.com/26196/eng/SKL_Gold_FSP.tgz

osresearch, 2016-10-08:
Several minor changes required for coreboot: diffs are not yet committed but were discussed on the mailing list.  https://ticket.coreboot.org/issues/77 and https://ticket.coreboot.org/issues/78

Xen required complete disabling of the vga console.  Even with `console=none` on the command line, it was hanging in `video_puts()`, possibly related to my patch to the Xen mailing list regarding the x230 VGA native initialization. https://patchwork.kernel.org/patch/9269155/

The coreboot payload kernel required framebuffer support to initialize the screen since there is no VGA bios.  This might also work on the x230; need to test it.

osresearch, 2016-10-08:
[image]

[image]

NodeGuy, 2017-02-02:
Hi, I love the work you're doing!  I want to buy a new machine to run Qubes OS and I'm excited about Heads.  Sorry to hijack this issue but can you confirm for me that Qubes OS works well on the [HP Chromebook 13](http://store.hp.com/us/en/ContentView?storeId=10151&catalogId=10051&eSpotName=chromebook13)?  How well is Heads working on it?  Is that the machine on the market today that you'd recommend the most or is there a better one aligned with our shared goals?

osresearch, 2017-02-02:
My summary of the Chell / G1:

* Heads + Qubes work fairly well on 8 GB RAM models, although 32 GB eMMC is slow and small.
* Battery life seems acceptable, 5-10 W power draw. Upgrading the dom0 Linux kernel might improve it.
* Hi-res screen is great (3200x1800), but it is like a high-gloss mirror.
* TPM works, although my meaured boot patches need porting to Chell mainboard.
* Keyboard and trackpad is one of the worst I've ever used.
* ACPI DMAR table is not yet implemented.
* S3 resume takes a really long time, not sure why.
* `kexec` hangs sometimes, not sure why.

[image]

Installing Heads right now requires soldering a debug port and a Servo debug board to be able to rewrite the entire boot ROM when things go wrong or need to be upgraded.  We might be able to do it from software once we have a rescue mode implemented in the Heads firmware.  This is probably the biggest roadblock to making the Chell a popular [Qubes R4 laptop](https://www.qubes-os.org/news/2016/07/21/new-hw-certification-for-q4/) since it meets almost all of the requirements.

NodeGuy, 2017-02-02:
Thank you so much for all of the detail!  That's a bummer about the keyboard and trackpad.  It's so easy to get spoiled by Apple hardware!

If I were to buy a Chell I think I'd reserve the 32 GB eMMC for dom0 and pair it with an [Atom RAID SSD](https://www.glyphtech.com/product/atom-raid-ssd) for general storage.

tlaurion, 2020-03-09:
@osresearch : reviewed board configs and Chell chromebook is not in boards? #691

closed 2024-04-16 by tlaurion