linuxboot/heads #39: Protected Regions and other flash protection platform features

zaolin, 2016-10-06:
Hey,

Maybe we want to implement them as well. Or extend the current implementation of coreboot itself. I know only PR is secure but I guess it would be good to have them in a security section with the BP feature.

osresearch, 2016-10-08:
The nonvolatile BP bits are definitely on the list of things to enable (issue #12), as well as experiments with disconnecting/grounding the #WP pin.

Figuring out where to set the PRR is wrapped up with the "how to do upgrades" issue.  If the BP bits sufficiently protect the bootblock and allow the hardware root of trust to be established, then it might be sufficient to leave the PRR unlocked in the recovery shell (in ROM).  Any updates would invalidate the TPM PCRs, allowing malicious changes to be detected.

osresearch, 2016-11-29:
The update strategy is also discussed in #17