linuxboot/heads #41: Trusted Boot

zaolin, 2016-10-06:
Hey,

in order to get the maximum sealing against the platform it would be useful to have a well documented and feature complete trusted boot in coreboot. I started to refactor the tpm stack and implement the missing features. Take a look at https://review.coreboot.org/#/q/status:open+tpm

osresearch, 2016-10-08:
Do we need more than a good root of trust as to the state of the ROM bootblock and the payload that we're going to launch?  The changes that I've made to coreboot (https://github.com/osresearch/coreboot/commit/033623b2d2e9d080e70136f6708f8467174b7a2c and others) are fairly minimal and allow most of the policy to be set in the Linux payload instead.

zaolin, 2016-10-08:
Normally yes. If you measure also blobs which are executed parts of the sb and cpu via intel txt then you get a better sealing for more hardware parts. So the attacker can't change for example cpu and sb. I want to make tpm spec for the measurements and do the integration into coreboot. TCPA ACPI log is also very useful if you want to pre-calculate changes after updates etc..

zaolin, 2016-10-19:
Timothy Pearson is also needing this for his TALOS workstation stuff. See https://www.crowdsupply.com/raptor-computing-systems/talos-secure-workstation

tlaurion, 2021-02-11:
@zaolin more riticisms around https://github.com/coreboot/coreboot/commit/c79e96b4eb310db9d44e36e2dff072c01469c380 ?