linuxboot/heads #42: Secure UI

zaolin, 2016-10-06:
Hey,

for an UI improvement I would recommend something like http://invisible-island.net/dialog/ . Which is really easy to configure and has an advanced feature set.

osresearch, 2016-10-08:
We'll need to install the ncurses library into the ROM as well to support it.  On the Chell chromebook (issue #38) that's no problem, although we are starting to get squeezed into the 4MB limit on the upper ROM on the x230.

[image]

I wonder if we should use a framebuffer library instead so that we can take advantage of larger fonts.  The 3200x1800 screen on the Chell means the default font is microscopic.

zaolin, 2016-10-08:
Okay we could set the font size dynamically for the screen resolution. Maybe we find a better UI library but that's not so easy. There are many libraries but they need mesa or x11.

zaolin, 2016-11-23:
Better idea we use plymouth here if possible. Which looks nicer does scaling itself etc.. So we need to write a plymouth theme here.

zaolin, 2016-11-24:
Plymouth theme is nearly ready except the design. https://github.com/zaolin/heads-plymouth-theme
@osresearch Today I am going to add all necessary stuff in the initrd.

zaolin, 2016-11-28:
See #50

tlaurion, 2019-05-21:
@zaolin : reread this old thread. Please reopen if whiptail and fbwhiptail doesn't fit.

closed 2019-05-21 by tlaurion