linuxboot/heads #60: Xen kernel not automatically built?

tlaurion, 2016-12-10:
Even though the xen references in Makefile are uncommented, the module is not compiled, patched and packaged.

git diff Makefile:
```diff --git a/Makefile b/Makefile
index ea0007c..4234165 100644
--- a/Makefile
+++ b/Makefile
@@ -148,7 +148,7 @@ $(foreach _, $(call libs,tpmtotp), $(eval $(call initrd_lib_add,$_)))
 $(foreach _, $(call libs,mbedtls), $(eval $(call initrd_lib_add,$_)))
 $(foreach _, $(call libs,qrencode), $(eval $(call initrd_lib_add,$_)))
 
-#$(foreach _, $(call outputs,xen), $(eval $(call initrd_bin,$_)))
+$(foreach _, $(call outputs,xen), $(eval $(call initrd_bin,$_)))
 
 # hack to install busybox into the initrd
 initrd_bins += initrd/bin/busybox
@@ -174,7 +174,7 @@ $(build)/$(coreboot_dir)/util/cbmem/cbmem: $(build)/$(coreboot_dir)/.canary
 #initrd_bins += initrd/bin/cryptsetup
 initrd/bin/cryptsetup: /sbin/cryptsetup
        cp "$<" "$@"
-initrd_bins += initrd/bin/dmsetup
+#initrd_bins += initrd/bin/dmsetup
 initrd/bin/dmsetup: /sbin/dmsetup
        cp "$<" "$@"
 initrd_bins += initrd/bin/gpgv
@@ -191,8 +191,6 @@ initrd_lib_install: $(initrd_bins) $(initrd_libs)
                initrd/bin/* \
                initrd/sbin/* \
 
-
-
 #
 # initrd image creation
 #
```

osresearch, 2016-12-10:
Oops. I removed it from the build list since the xen makefiles take a while to realize that there is nothing to do, so I didn't want it built on every `make` invocation.

It doesn't fit in the 4 MB initrd, so it shouldn't be in the `initrd_bins` list, but it does need to be built at least once for installing Qubes.

tlaurion, 2016-12-12:
Where should it be? Referred Makefile doesn't create it.

tlaurion, 2016-12-16:
Simple enough:
```
diff --git a/Makefile b/Makefile
index ba51138..2fb3982 100644
--- a/Makefile
+++ b/Makefile
@@ -20,7 +20,7 @@ include modules/*
 
 # These will be built via their intermediate targets
 # This increases the build time, so it is commented out for now
-#all: $(foreach m,$(modules),$m.intermediate)
+all: $(foreach m,$(modules),$m.intermediate)
 
 define prefix =
 $(foreach _, $2, $1$_)

```

closed 2016-12-16 by tlaurion