linuxboot/heads #68: Build of x230 coreboot with 4.5 fails on first build

osresearch, 2016-12-13:
```
    OBJCOPY    ramstage/cpu/x86/smm/smm.manual
    CC         generated/ramstage.o
x230/romstage/northbridge/intel/sandybridge/romstage.o: In function `mainboard_romstage_entry':
/build/heads/build/coreboot-git/src/northbridge/intel/sandybridge/romstage.c:74: undefined reference to `tlcl_measure'
/build/heads/build/coreboot-git/src/northbridge/intel/sandybridge/romstage.c:77: undefined reference to `tlcl_measure'
x230/romstage/northbridge/intel/sandybridge/romstage.o: In function `platform_segment_loaded':
/build/heads/build/coreboot-git/src/northbridge/intel/sandybridge/romstage.c:149: undefined reference to `tlcl_measure'
src/arch/x86/Makefile.inc:264: recipe for target 'x230/cbfs/fallback/romstage.debug' failed
make[1]: *** [x230/cbfs/fallback/romstage.debug] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/build/heads/build/coreboot-git'
Makefile:108: recipe for target 'coreboot.intermediate' failed
make: *** [coreboot.intermediate] Error 2
```

Re-running the build works fine:

```
    OBJCOPY    cbfs/fallback/romstage.elf
    CC         cbfs/fallback/ramstage.debug
Performing operation on 'COREBOOT' region...
Created CBFS (capacity = 4194008 bytes)
Performing operation on 'COREBOOT' region...
Performing operation on 'COREBOOT' region...
    CBFS       fallback/romstage
Performing operation on 'COREBOOT' region...
E: Input file size (104188) greater than page size (65536).
    CBFS       mrc.cache
Performing operation on 'COREBOOT' region...
    CBFS       cpu_microcode_blob.bin
Performing operation on 'COREBOOT' region...
    CBFS       fallback/ramstage
Performing operation on 'COREBOOT' region...
    CBFS       cmos.default
Performing operation on 'COREBOOT' region...
    CBFS       cmos_layout.bin
Performing operation on 'COREBOOT' region...
    CBFS       fallback/dsdt.aml
Performing operation on 'COREBOOT' region...
    CBFS       fallback/payload
Performing operation on 'COREBOOT' region...
E: The stage file is not in ELF format!

** WARNING **
Coreboot will be built with a fake Intel Firmware Descriptor (IFD).
Never write a complete coreboot.rom with a fake IFD to your board's
flash ROM! Make sure that you only write valid flash regions.

    IFDFAKE    Building a fake Intel Firmware Descriptor
    DD         Adding Intel Firmware Descriptor
    CBFS       coreboot.rom
    CBFSPRINT  coreboot.rom

Performing operation on 'COREBOOT' region...
Name                           Offset     Type         Size
cbfs master header             0x0        cbfs header  32
cpu_microcode_blob.bin         0x80       microcode    22528
cmos.default                   0x5900     cmos_default 256
cmos_layout.bin                0x5a40     cmos_layout  1952
fallback/dsdt.aml              0x6240     raw          13847
(empty)                        0x98c0     null         26264
fallback/romstage              0xff80     stage        79172
(empty)                        0x23540    null         51544
mrc.cache                      0x2fec0    mrc_cache    65536
fallback/ramstage              0x3ff00    stage        91332
fallback/payload               0x56400    payload      3733681
(empty)                        0x3e5d00   null         104920
bootblock                      0x3ff700   bootblock    1952

Built lenovo/x230 (ThinkPad X230)
```

Perhaps related to `make -j 8`?

zaolin, 2016-12-15:
I fixed the tpm acpi issue https://review.coreboot.org/#/c/17855/ . I would recommend to tag the source that we don't have similar issues later again.

mfc, 2016-12-15:
for me it's failing at:
```
checking uuid/uuid.h usability... no
checking uuid/uuid.h presence... no
checking for uuid/uuid.h... no
configure: error: You need the uuid library.
Makefile:108: recipe for target '/home/user/heads-master/build/cryptsetup-1.7.3/.configured' failed
make: *** [/home/user/heads-master/build/cryptsetup-1.7.3/.configured] Error 1
```
re-running `make` doesn't change anything.

osresearch, 2016-12-16:
@mfc - to build `cryptsetup` you'll need quite a few extra packages on your build system (like `uuid-devel`).  We're not entirely self-hosting in that respect, which will be a problem for actual reproducible builds.

mfc, 2016-12-16:
yeah getting that error with `uuid-devel` installed. anyway may be a different issue, didn't mean to hijack this issue.

osresearch, 2017-03-01:
I think this is no longer an issue? I did a clean checkout to test `Makefile` updates (for issues #114 and #111) and the make ran to completion.

closed 2017-03-01 by osresearch