linuxboot/heads #69: S3 resume script in coreboot?

osresearch, 2016-12-18:
Does coreboot use something like the SMM lockbox to prevent S3 script hijacking? This won't matter as much if #12 (SPI HW write protection) is fixed, but until then the system is potentially vulnerable to code execution during resume.  Even if FLOCKDN is set early enough, it is a powerful place for malware to hide.

osresearch, 2016-12-18:
We've already made `LOCK_SPI_ON_RESUME_RO` the default in the Heads coreboot config file, but need to validate when it is being setup during the resume process. https://www.coreboot.org/pipermail/coreboot/2016-April/081224.html

osresearch, 2016-12-18:
Not sure if this is locked on the Chell -- the config parameter only appears in coreboot/src/southbridge/intel/bd82x6x/finalize.c, so we might be depending on the Intel FSP to do the locking (cue repeat of Snorlax/Prince Harming).

osresearch, 2016-12-18:
Really nice overview of the S3 sleep/resume process including the interaction of AML, the kernel and the firmware. https://wiki.ubuntu.com/Kernel/Reference/S3

Siproqu, 2020-07-16:
Is this still a thing? Shouldn't `LOCK_SPI_ON_RESUME_NO_ACCESS` or `LOCK_SPI_ON_RESUME_RO` be set as default?

tlaurion, 2022-05-25:
The platform lockdown configuration in coreboot changed a lot since this issue was opened.
The new default should become what is under #326, which would be the safer default for Sandy/Ivy bridge, but where FSP workarounds need to be pushed upstream to be able to deal properly with platform lockdown.

Discussion happened under https://github.com/tlaurion/heads/commit/3343f8dcbbc608043f32a608ce85e746bba51412 for coreboot 4.13+

Basically requiring io386 module and then the following addition under coreboot config:
- `CONFIG_BOOTMEDIA_LOCK_CONTROLLER=y`
- ` # CONFIG_INTEL_CHIPSET_LOCKDOWN is not set`

And then:
- modification of kexec-boot script to call `lock_chip` just prior of [kexec call](https://github.com/tlaurion/heads/commit/3343f8dcbbc608043f32a608ce85e746bba51412#diff-5a9e3f59233a0922d264a00f00e94ba14dab66f0cef5a43029bd86f3d073038aR130-R133)

@root-hardenedvault added FSP hacks on his Heads fork for newer platforms.

closed 2025-02-13 by tlaurion