linuxboot/heads #77: Use musl-libc instead of system libc

osresearch, 2016-12-29:
Building with alpine linux didn't work well (issue #54), but it appears that it is easier to do a hybrid build of linux and coreboot with the regular gcc, and the initrd tools with the `musl-gcc` wrapper.

rofl0r, 2016-12-30:
you might as well try sabotage linux https://github.com/sabotage-linux/sabotage .
our current initrd is 1.3 MB with full busybox, cryptsetup and lvm (everything static linked).

Kokokokoka, 2016-12-30:
And what about buildroot? It is used by jeltka, coreboot payload, one can choose between uclibc-ng, musl,
there are external projects like https://github.com/xSmurf/coreboot_buildroot_project I'm trying to add zfs
package here as zfs will soon have native encryption and I'll try to use it with TPM and my payload will
be small busybox+zfs-tools+xen, then I'll install Qubes, now I'm using buildroot with patches from this
repo, and it works fine.

osresearch, 2016-12-30:
Buildroot looks promising, especially since it includes (experimental) support for reproducible builds.One concern is that it doesn't seem to fit -- xSmurf says "_Installing on the X220 requires hardware modification as the flash has to be upgraded to 16MiB._"  With the Heads bundler the entire runtime fits into the 4MB of the top SPI flash (and my experimental musl-libc branch has 400 KB free).

rofl0r, 2016-12-30:
since you don't mention sabotage just a short heads-up:
- sabotage was the first musl distro (since 2011)
- i (sabotage's main author) am the top 3 contributor to musl libc in number of commits and top 1 contributor in number of bugs reported and also top 1 contributor in number of upstream bugs reported.
- all the other musl-based distros and frameworks like buildroot, openwrt etc just jumped on to the wagon years later. without the work done on sabotage there would probably be no musl-based alpine-linux, etc.
- it should be perfect for your use-case: you just build stage0 according to the readme, which takes ~2 min and you already got a system you can chroot into and build further stuff.

Kokokokoka, 2016-12-30:
> Buildroot looks promising, especially since it includes (experimental) support for reproducible builds.One concern is that it doesn't seem to fit -- xSmurf says "Installing on the X220 requires [...]
This assumption is wrong, I'm using buildroot, and I've got 1.6 mb payload, with wpa-supplicant and dropbear and me 1.5mb. (I'll tell what is the real size of the payload later)
I'll share my config if it's needed, I don't remember for sure, but there was plenty of the free space with
bloated config there was about 200 kb free.

osresearch, 2017-01-04:
Thanks for the review on the musl-libc branch merge.  sabotage looks intriguing as a future platform; I had run across it earlier as a way to build busybox with musl-libc and we're using its kernel headers package to host the gcc wrapper.

The changes are now in master and appear to build from a clean checkout (on ubuntu and fedora). This doesn't satisfy the reproducible build goal yet (issue #2), but it does close out dependency on the system libc (issue #7).

closed 2017-01-04 by osresearch