linuxboot/heads #82: PGP key has expired

Tsutsukakushi, 2017-01-19:
[Your PGP key](https://trmm.net/PGP) has expired on 14th of November 2016.

I tried to email you about this but you probably haven't noticed.

Do you have any plans to generate a new one?

RussellSenior, 2017-01-19:
@Tsutsukakushi fwiw, the key as currently published on pgp.mit.edu has an expiration date of 2017-10-06.

osresearch, 2017-02-01:
Thanks for noticing the expired key.  I've replaced it with the updated version.

Although users should replace the key with their own -- I can't sign your hypervisor or root hashes, so it is important that *your* public key be in the initramfs.  The instructions for doing so need to be in the `README.md` file.

Longer term, I'd like to move the keys out of the `initrd.cpio` file so that the coreboot, kernel and initrd are identical for every build for every user.  This could be done by adding a cbfs file with the user public keys, and the `init` script could measure and unpack that extra filesystem.

closed 2017-02-01 by osresearch

Tsutsukakushi, 2017-02-01:
It's also important for verifying the integrity of the git repo, not just for boot.

Tsutsukakushi, 2017-02-01:
And you still have the expired key up on your site, which is the one you use to sign commits.

RussellSenior, 2017-02-11:
@osresearch this link still shows the key as expired: https://trmm.net/PGP

Tsutsukakushi, 2017-03-20:
```
pub   4096R/0DEF86C0 2010-09-12 [expires: 2016-10-04]
      Key fingerprint = A6C7 4E34 1054 A169 CE52  BE5F B65B FE54 0DEF 86C0
uid                  Trammell Hudson <trammell.hudson@gmail.com>
uid                  Trammell Hudson <hudson@trmm.net>
uid                  Trammell Hudson <hudson@swcp.com>
uid                  Trammell Hudson <hudson@osresearch.net>
sub   2048R/DDECBE68 2014-11-15 [expires: 2016-11-14]
sub   2048R/71ACDB80 2014-11-15 [expires: 2016-11-14]
```

When?

osresearch, 2017-03-20:
Fixed.