linuxboot/heads #84: xen-4.6.3.gz is not automatically built anymore

sledge77, 2017-01-23:
For some reason it seems that currently the Makefile does not process the xen module file at all and consequently does not download the package to apply the heads specific patch. That leaves me with a laptop that runs Heads but cannot actually start Xen via kexec -e.

osresearch, 2017-01-23:
Typically the `xen-4.6.3.gz` is too large to fit into the initrd, so installing it is "left as an exercise for the reader".  As mentioned in issue #27, copying the modified xen into the Qubes `/boot` requires a manual step to move it to a USB device, mount that and the Qubes `/boot`, then copy it over.

This will require some cleanup in the `Makefile` to ensure that it is built automatically.  Until then, you can download, verify and patch Xen by running:

```
make xen.intermediate
```

sledge77, 2017-01-24:
Ahh, ok. The need to use the intermediate target was non-obvious to me.
It builds the xen.gz file now which I rename to xen-4.6.3.gz and put it on the /boot mount point of the internal hard disk as described.

The call to "kexec -l" now fails with a "Cannot determine the file type of /boot/xen-4.6.3.gz". Any ideas?

osresearch, 2017-01-24:
Yes, oops.  `kexec` is being built without gzip support.  You can temporarily work around it by `gunzip`ing `xen.gz` and then `kexec -l xen` instead.

osresearch, 2017-01-24:
https://github.com/osresearch/heads/commit/5fd9878d28cd05f89551359cc0de1adfb67a310b added `--without-zlib` when I was removing all the extra dependencies that we were pulling in from the host's environment.  `busybox` has `gunzip`, and we'll end up with a duplicate implementation in `libz`.

osresearch, 2017-04-04:
Xen is now built and included in the `x230.rom` to make it easier to bootstrap Qubes (issue #27 )

closed 2017-04-04 by osresearch