linuxboot/heads #85: Testing On A New Machine

drvdevd, 2017-01-24:
I would like to help cobtribute to this project and I have a few machines to play with.

First of all, I have a Toshiba Chromebook 2 (aka "swanky") on which I've already flashed one of John Lewis' custom SeaBIOS inplementations (see: https://johnlewis.ie/custom-chromebook-firmware/rom-download/), using the 'flashrom' utility from within Chrome OS.

Do you think it would it be possible to flash heads using that utility and is there interest in supporting more than the "core" platforms (seemingly: the Thinkpad and the newer chromebook model)?

Thanks!

osresearch, 2017-01-24:
The Swanky has a N2840 CPU, which has VT-x but not VT-d.  Most of the focus has been on machines that are a good fit for the [Qubes 4.0 HCL](https://www.qubes-os.org/news/2016/07/21/new-hw-certification-for-q4/), which requires an IOMMU.  Which OS are you planning to kexec?

drvdevd, 2017-01-24:
I would probably only want to run vanilla Linux, 3.x - 4.x. FreeBSD is on my radar but Im not sure how many peripherals it supports on this machine. I am curious if it would be possible to kexec it though.

closed 2025-02-13 by tlaurion