linuxboot/heads #86: Testing with nested virt and emulated TPM

drvdevd, 2017-01-24:
Has any attempt been made to do this: test heads in KVM with nested virt and an emulated TPM?

Not sure if this has landed in QEMU/KVM yet, but I will research and update this issue. Some more background:

http://wiki.qemu.org/Features/TPM

drvdevd, 2017-01-24:
Ok, it would appear Stefan Berger has open sourced at least some part of a qemu fork with TPM support:

https://github.com/stefanberger/qemu-tpm/commits/v2.8.0%2Btpm

I will give this a try and report back.

osresearch, 2017-01-24:
I've had trouble starting Xen under Qemu (it starts up, but triple faults when starting the dom0 kernel?) and haven't debugged the nested virtualization yet.  If you get it working, I'd be eager to know what needs to be done.

tlaurion, 2019-05-21:
@drvdevd any updates on this?

tlaurion, 2024-04-16:
qemu-coreboot-* boards do exacrly that.

closed 2024-04-16 by tlaurion