linuxboot/heads #87: Thinkpad x230 EC firmware

osresearch, 2017-01-24:
At linux.conf.au 2017, Hamish Coleman showed how to reflash the Thinkpad EC with custom firmware; this would be a nice step to eliminate a remaining piece of vendor code.  Video: https://www.youtube.com/watch?v=Fzmm87oVQ6c

merge, 2018-05-23:
FWIW, hamish's projects aren't about eliminating vendor code. It's about changing as little as possible in binaries to achieve certain changes. OTOH he tries to simplify the installation process, especially with https://github.com/hamishcoleman/thinkpad-dosflash which is a work in progress.

so currently there's a bootable image capable of flashing modified EC firmware. Nothing that really could be nicely integrated with Heads nor eliminate vendor code...

tlaurion, 2020-08-01:
There was confusion in this.

closed 2020-08-01 by tlaurion