linuxboot/heads #98: Bootstrap Issue with GCC 6.3.0

blackwellops, 2017-01-30:
I'm not sure if this "issue" should be here or on the [musl-cross](https://bitbucket.org/GregorR/musl-cross) repo. 

However, when trying to `./bootstrap` with GCC 6.3.0 installed. I received this error. 

```shell
In file included from ../../gcc/cp/except.c:1023:0:
cfns.gperf: In function ‘const char* libc_name_p(const char*, unsigned int)’:
cfns.gperf:101:1: error: ‘const char* libc_name_p(const char*, unsigned int)’ redeclared inline with ‘gnu_inline’ attribute
cfns.gperf:26:14: note: ‘const char* libc_name_p(const char*, unsigned int)’ previously declared here
g++ -c  -DIN_GCC_FRONTEND  -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -Icp -I../../gcc -I../../gcc/cp -I../../gcc/../include -I../../gcc/../libcpp/include -I/home/***/git/heads/build/musl-cross/gcc-5.3.0/build2/./gmp -I/home/***/git/heads/build/musl-cross/gcc-5.3.0/gmp -I/home/***/git/heads/build/musl-cross/gcc-5.3.0/build2/./mpfr/src -I/home/***/git/heads/build/musl-cross/gcc-5.3.0/mpfr/src -I/home/***/git/heads/build/musl-cross/gcc-5.3.0/mpc/src  -I../../gcc/../libdecnumber -I../../gcc/../libdecnumber/bid -I../libdecnumber -I../../gcc/../libbacktrace   -o cp/init.o -MT cp/init.o -MMD -MP -MF cp/.deps/init.TPo ../../gcc/cp/init.c
cfns.gperf: At global scope:
cfns.gperf:26:14: warning: inline function ‘const char* libc_name_p(const char*, unsigned int)’ used but never defined
Makefile:1065: recipe for target 'cp/except.o' failed
make[2]: *** [cp/except.o] Error 1
```

Which seems to be related to this [https://gcc.gnu.org/bugzilla/show_bug.cgi?id=69959](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=69959). 

Another example  [https://github.com/LibreELEC/LibreELEC.tv/pull/384](https://github.com/LibreELEC/LibreELEC.tv/pull/384) recommends using ``` export CXXFLAGS="$CXXFLAGS -std=gnu++98"```to remedy this. 

However, I do not know what impact that has. 

I recreated this on Debian Sid and in a docker container using `debian:stretch`. 

Installing clang instead of gcc seemed to resolve this.

rofl0r, 2017-01-30:
i've fixed that issue just yesterday in musl-cross https://github.com/GregorR/musl-cross/commit/76015715836ec684d1218d7aa63cde0c0e515f4d (at least for gcc 4.7.4). which gcc version are you trying to build ? also note that the bitbucket repo is unmaintained.

blackwellops, 2017-01-30:
gcc-5.3.0

rofl0r, 2017-01-31:
fixed. https://github.com/GregorR/musl-cross/commit/e01e535590fd6e0d2c1c289d990a2fcae1215071

closed 2017-01-31 by blackwellops