linuxboot/heads #105: TPM PCR rewrites might fail to badfet attacks

osresearch, 2017-02-08:
[image]

The TPM libraries don't necessarily make sure that the values written to the PCRs are correctly recorded. A potential vulnerability exists if a badfet style device can interfere with the low-speed LPC bus. One approach is to read the values back and ensure that they match the expected value, although this is still potentially vulnerable.

jonathanfemideer, 2017-02-27:
Can you link to some documentation about this?

osresearch, 2017-02-27:
They don't have a video posted, unfortunately.  Ang and Rick from Red Balloon Security presented their preliminary research at Empire Hacking earlier this month: https://www.empirehacking.nyc/archive/

One hypothetical scenario is that we want ensure the PCRs are reset when we start a recovery shell (issue #16) or after the tpmtotp retrieves the shared secret (issue #25). If an attacker can prevent the LPC writes that extend the PCRs during this step, they will be able to unseal secrets from the recovery shell.

jpouellet, 2017-04-04:
Apologies if this question is uninformed, as I have not read the TCG specs.

If someone is probing the LPC bus, what prevents them from performing a clean boot while sniffing & logging everything, powering off & replaying the measurements, then booting their own payload while DoSing the bus, and arriving in a state where they have the PCRs of a legit boot, but with their own code running instead?

If the above is indeed the attack, then I don't see how reading back values would help since the adversary does not need to inject faults while Heads is the thing booting in order to pull off the attack.

osresearch, 2017-04-04:
If the adversary has access to the LPC bus they can directly play measurements into the TPM and have it unseal secrets.  What they can't do is decrypt the disk encryption blob (which is sealed and encrypted), so this doesn't give them the keys.  An attacker with this capability is very tough to defeat since they could just swap the flash chip with their own.

The concern about the red balloon research is that they don't need access to the internals and can perform their attack on an unopened machine.  It is still a very hypothetical and far fetched, but worth thinking about how to defend against.

jpouellet, 2017-04-05:
> The concern about the red balloon research is that they don't need access to the internals and can perform their attack on an unopened machine. It is still a very hypothetical and far fetched, but worth thinking about how to defend against.

Woah, magic! I'd definitely be interested in seeing/reading their talk/paper if/when you hear of a published copy.