linuxboot/heads #106: latest version fails during musl-cross-git build

mfc, 2017-02-24:
after running `make` on freshly downloaded `heads-master`:
```
git clone "https://github.com/GregorR/musl-cross" "/home/user/Documents/heads-master/build/musl-cross-git"
Cloning into '/home/user/Documents/heads-master/build/musl-cross-git'...
remote: Counting objects: 746, done.
remote: Total 746 (delta 0), reused 0 (delta 0), pack-reused 746
Receiving objects: 100% (746/746), 278.20 KiB | 256.00 KiB/s, done.
Resolving deltas: 100% (469/469), done.
Checking connectivity... done.
if [ -r patches/musl-cross.patch ]; then ( cd /home/user/Documents/heads-master/build/musl-cross-git ; patch -p1 ) < patches/musl-cross.patch; fi
patching file config.sh
touch "/home/user/Documents/heads-master/build/musl-cross-git/.canary"
touch "/home/user/Documents/heads-master/build/musl-cross-git/.config"
cd "/home/user/Documents/heads-master/build/musl-cross-git" ; echo > Makefile -n '/home/user/Documents/heads-master/crossgcc/x86_64-linux-musl/bin/x86_64-linux-musl-gcc:\n\tCC_BASE_PREFIX="/home/user/Documents/heads-master/crossgcc" ./build.sh'
touch "/home/user/Documents/heads-master/build/musl-cross-git/.configured"
/home/user/Documents/heads-master/build/make-4.2/make -C "/home/user/Documents/heads-master/build/musl-cross-git" 
make[2]: Entering directory '/home/user/Documents/heads-master/build/musl-cross-git'
make[2]: *** No targets.  Stop.
make[2]: Leaving directory '/home/user/Documents/heads-master/build/musl-cross-git'
make[1]: *** [Makefile:138: musl-cross.intermediate] Error 2
make[1]: Leaving directory '/home/user/Documents/heads-master'
Makefile:300: recipe for target 'all' failed
make: *** [all] Error 2
```

happens on Fedora 23, Fedora 24, and Debian 8.

osresearch, 2017-02-26:
I think I see the problem.  The `echo -n` should be `echo -e` to get escape character expansion.  Try this patch:

```
diff --git a/modules/musl-cross b/modules/musl-cross
index a1814df..4626e2a 100644
--- a/modules/musl-cross
+++ b/modules/musl-cross
@@ -7,7 +7,7 @@ musl-cross_repo := https://github.com/GregorR/musl-cross
 musl-cross_output := $(pwd)/crossgcc/x86_64-linux-musl/bin/x86_64-linux-musl-gcc
 
 musl-cross_configure := \
-	echo > Makefile -n \
+       echo > Makefile -e \
	'$(musl-cross_output):\n\tCC_BASE_PREFIX="$(pwd)/crossgcc" ./build.sh'
 
 musl-cross_target :=
```

mfc, 2017-02-26:
thanks, fixes the problem!

mfc, 2017-02-26:
sorry will let you close it ;)

osresearch, 2017-02-26:
Patched!  Thanks for testing it.

closed 2017-02-26 by osresearch