linuxboot/heads #107: Locale issue: FAT-fs (sdb2): IO charset iso8859-1 not found

jonathanfemideer, 2017-02-24:
## Background

I have successfully reached this point of the [*Installing Heads*](https://trmm.net/Installing_Heads) instructions: "Congratulations! You now have a Coreboot + Heads Linux machine."

Thank you for getting me this far =)

I want to skip, for now, configuring the PGP key and the TPM, and go straight to the "Installing Qubes" section of *Installing Heads*. That is because if I can get Heads to boot Qubes then at least I will have a working X230 running a libre BIOS/UEFI, and a reasonably secure operating system. I can always reinstall Qubes from scratch when I get around to configuring the PGP key and the TPM.

## Getting Heads to boot Qubes

First, an aside: is Heads expecting the current stable release of Qubes, i.e. [Qubes 3.2](https://www.qubes-os.org/downloads/)? Installing Qubes 3.2 from DVD to a 2.5" 7mm SSD on a spare, non-Heads PC, yields, among other things, `/boot/xen-4.6.*1*.gz`. However, Heads's [`start-xen`](https://github.com/osresearch/heads/blob/master/initrd/bin/start-xen#L6) script, and the  "[Installing Qubes](https://trmm.net/Installing_Heads#Installing_Qubes)" section of *Installing Heads* clearly show that Heads is expecting `/boot/xen-4.6.*3*.gz` .

Anyhow, following the instructions at https://trmm.net/Installing_Heads#Installing_Qubes goes fine until this step:

    # mount -o ro /dev/sdb2 /boot

This gives:

    [7522.527845] FAT-fs (sdb2): IO charset iso8859-1 not found

This is a surprising warning, given that the Qubes ISO should be identical, wherever in the world it was downloaded. In other words, @osresearch's copy of the Qubes 3.2 ISO should be the same of mine. I did check the signature on mine. Also, I built Heads on a machine that has a UTF-8 based locale.

## Questions

The drive does get mounted, but is it safe to boot?

And how might I avoid the warning above? Presumably there's a configuration setting that has caused the minimal Linux kernel within Heads to be built without [ISO 8559-1](https://en.wikipedia.org/wiki/ISO_8859-1) support.

jonathanfemideer, 2017-02-24:
> The drive does get mounted, but is it safe to boot?

I tried booting it. There were no obvious problems.

> First, an aside: is Heads expecting the current stable release of Qubes, i.e. Qubes 3.2? Installing Qubes 3.2 from DVD to a 2.5" 7mm SSD on a spare, non-Heads PC, yields, among other things, [...]

AFAICT, the *Installing Heads* documentation was written when Qubes 3.2-rc3 was the recommended version of Qubes. That still doesn't explain why Heads uses Xen 4.6.3, though.

osresearch, 2017-02-25:
Thanks for the super detailed bug reports, @jonathanfemideer.

The charset issue is spurious, although it does mess up upper/lowercase handling on FAT filesystems.  We should include the kernel module for it -- during the quest to reduce the size of the kernel everything extra was disabled.

jonathanfemideer, 2017-02-26:
Thanks for making Heads, and for responding promptly to bug reports. You rock!

osresearch, 2017-03-20:
iso8559-1 is now enabled in the kernel.

closed 2017-03-20 by osresearch