linuxboot/heads #111: My build isn't reproducible

JohnnyLeone, 2017-02-25:
I'm trying to build [v0.1.0](https://github.com/osresearch/heads/releases/tag/v0.1.0) with docker on a arch linux host. As docker image i'm using debian:jessie and ubuntu:16.04 both with the same hash of the resulting binary.

Actually the build process goes pretty streight forward to the resulting x230.rom binary but the hash doesn't match the hash for this release.
[image]

What i did is, i only edited the Makefile to build for my x230 target.

Also if i flash the resulting binary to my x230 booting ends up with a kernel panic.
[image]

osresearch, 2017-02-26:
It looks like the `initrd.cpio` might not be correct.  Can you verify its contents by running `cpio -tv < initrd.cpio`?

JohnnyLeone, 2017-02-26:
Here is the output of `cpio -tv < initrd.cpio`:
```
root@3a74cccbc369:/heads/heads-0.1.0# cpio -tv < initrd.cpio 
drwx------   0 root     root            0 Jan  1  1970 .
-rw-------   0 root     root          110 Jan  1  1970 .ash_history
drwx------   0 root     root            0 Jan  1  1970 .gnupg
-rw-------   0 root     root        25358 Jan  1  1970 .gnupg/trustedkeys.gpg
drwx------   0 root     root            0 Jan  1  1970 /dev
crw-------   0 root     root     136,   3 Jan  1  1970 /dev/console
drwx------   0 root     root            0 Jan  1  1970 bin
lrwx------   0 root     root            7 Jan  1  1970 bin/[ -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/[[ -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/ash -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/awk -> busybox
-rwx------   0 root     root         6832 Jan  1  1970 bin/base32
lrwx------   0 root     root            7 Jan  1  1970 bin/base64 -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/basename -> busybox
-rwx------   0 root     root       336960 Jan  1  1970 bin/busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/bzip2 -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/cat -> busybox
-rwx------   0 root     root        19888 Jan  1  1970 bin/cbmem
lrwx------   0 root     root            7 Jan  1  1970 bin/chattr -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/chmod -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/cp -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/cpio -> busybox
-rwx------   0 root     root        55560 Jan  1  1970 bin/cryptsetup
lrwx------   0 root     root            7 Jan  1  1970 bin/cut -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/date -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/dc -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/dd -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/df -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/dirname -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/dmesg -> busybox
-rwx------   0 root     root       133216 Jan  1  1970 bin/dmsetup
lrwx------   0 root     root            7 Jan  1  1970 bin/echo -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/egrep -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/env -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/expr -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/false -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/fgrep -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/find -> busybox
-rwx------   0 root     root          338 Jan  1  1970 bin/generate-crypttab
lrwx------   0 root     root            7 Jan  1  1970 bin/getopt -> busybox
-rwx------   0 root     root       418000 Jan  1  1970 bin/gpgv
lrwx------   0 root     root            7 Jan  1  1970 bin/grep -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/groups -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/gunzip -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/gzip -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/hd -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/head -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/hexdump -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/hostid -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/id -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/install -> busybox
-rwx------   0 root     root       158704 Jan  1  1970 bin/kexec
lrwx------   0 root     root            7 Jan  1  1970 bin/kill -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/killall -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/less -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/ln -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/ls -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/lsattr -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/lsof -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/lspci -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/lsusb -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/md5sum -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/mkdir -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/mkfifo -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/mknod -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/more -> busybox
lrwx------   0 root     root            7 Jan  1  1970 bin/mount -> busybox
[... 55 lines cut ...]
-rwx------   0 root     root       332504 Jan  1  1970 lib/libtpm.so
-rwx------   0 root     root        11624 Jan  1  1970 lib/libuuid.so.1
-rwx------   0 root     root         1373 Jan  1  1970 mount-boot
drwx------   0 root     root            0 Jan  1  1970 sbin
lrwx------   0 root     root           14 Jan  1  1970 sbin/blkid -> ../bin/busybox
lrwx------   0 root     root           14 Jan  1  1970 sbin/chroot -> ../bin/busybox
[... cut ...]
lrwx------   0 root     root           14 Jan  1  1970 sbin/lsmod -> ../bin/busybox
lrwx------   0 root     root           14 Jan  1  1970 sbin/modinfo -> ../bin/busybox
lrwx------   0 root     root           14 Jan  1  1970 sbin/modprobe -> ../bin/busybox
lrwx------   0 root     root           14 Jan  1  1970 sbin/powertop -> ../bin/busybox
lrwx------   0 root     root           14 Jan  1  1970 sbin/rmmod -> ../bin/busybox
lrwx------   0 root     root           14 Jan  1  1970 sbin/sysctl -> ../bin/busybox
6051 blocks
```

JohnnyLeone, 2017-02-26:
Short update: i've sucessfully reproducible build the x230.rom with a ubuntu 16.04 virtual machine.

osresearch, 2017-02-26:
At a quick glance it looks like the `/dev/console` in the container is being virtualized in some way, so it has the wrong device major/minor.  I'm going to rename this issue to reflect that problem so that we can think of how to create the device file independent of the build system (and without root access to run `mknod`).

osresearch, 2017-02-28:
I've just pushed a change that has a pre-built `/dev/console` initrd. @JohnnyLeone - Can you try building in the container to see if it produces a reproducible, bootable image?

JohnnyLeone, 2017-02-28:
I've build heads against 26b323d in a debian:jessie container. It builds successfully and also boots without kernel panic on my x230.

The resulting SHA256 Hash is:
`0a33baf145f4b2f23b0d047fa7ec556f4c70a921cbbdd9aba665421e2a128336  x230.rom`

osresearch, 2017-03-01:
Is that a clean build or did you install anything in the initrd?  The [0.1.0 x230.rom hash](https://github.com/osresearch/heads/releases/tag/v0.1.0) should be:

```
83df0b6845fdd6b335119f7e89e08c47ec3566f515fe93bc528fcd955587a43e  x230.rom
```

(and I believe a build from `master` should still have the same result, since we've only tweaked some of the makefiles and non-code related things since then)

JohnnyLeone, 2017-03-01:
It was a complete clean build, now i have verified the result with another build in a ubuntu:16.04 container. With the same resulting sha256 hash. But it seems still that the fallback/payload has another size in the container one.

That's the output of the reproducible build:
```
Name                           Offset     Type         Size
cbfs master header             0x0        cbfs header  32
cpu_microcode_blob.bin         0x80       microcode    22528
cmos.default                   0x5900     cmos_default 256
cmos_layout.bin                0x5a40     cmos_layout  1952
fallback/dsdt.aml              0x6240     raw          13847
(empty)                        0x98c0     null         26264
fallback/romstage              0xff80     stage        78116
(empty)                        0x23140    null         52568
mrc.cache                      0x2fec0    mrc_cache    65536
fallback/ramstage              0x3ff00    stage        85923
fallback/payload               0x54f00    payload      3306561
(empty)                        0x37c380   null         537432
bootblock                      0x3ff700   bootblock    1952
```

That's the output of the container build:
```
Name                           Offset     Type         Size
cbfs master header             0x0        cbfs header  32
cpu_microcode_blob.bin         0x80       microcode    22528
cmos.default                   0x5900     cmos_default 256
cmos_layout.bin                0x5a40     cmos_layout  1952
fallback/dsdt.aml              0x6240     raw          13847
(empty)                        0x98c0     null         26264
fallback/romstage              0xff80     stage        78116
(empty)                        0x23140    null         52568
mrc.cache                      0x2fec0    mrc_cache    65536
fallback/ramstage              0x3ff00    stage        85923
fallback/payload               0x54f00    payload      3306097
(empty)                        0x37c1c0   null         537880
bootblock                      0x3ff700   bootblock    1952
```
I've also compared the outupt of `cpio -tv < initrd.cpio`, it seems fixed until the dates are different. While the VM one shows `Dec 31  1969` the container one shows `Jan  1  1970`. 

And btw. yes the hash against commit 26b323d doesn't change.

osresearch, 2017-03-01:
The 1969 vs 1970 is likely a timezone thing on cpio display (the VM is in local time, the container is in UTC?).  Let's try to isolate which files are different -- can you run:

```
find heads/initrd -type f -print0 | xargs -0 sha256sum | sort
```

on the two build directories and diff the output?

JohnnyLeone, 2017-03-01:
You're right the VM is in local time and the container in UTC.

The only diff of both outputs was the gpgv binary
```
VM: 3cb885da20eac8413d78fbcf0e941b64e4d13b76d271b271ffcca709e5ec6d83  initrd/bin/gpgv
Container: 4705774e1d3814d3ba648faddbb97dcedbfad1607f515ded73405e3cea5c05a4  initrd/bin/gpgv
```

osresearch, 2017-03-31:
Is your `gpgv` executable still being build non-reproducibly with a current head?

osresearch, 2017-04-06:
Please re-open this if it is still an issue.  I've made a few change to the build process, documented here: http://osresearch.net/Building

closed 2017-04-06 by osresearch