linuxboot/heads #113: Heads documentation is hard to update

jonathanfemideer, 2017-02-26:
At the moment, the most extensive guide to Heads installation is the *Installing Heads* page on Trammell Hudson's website: https://trmm.net/Installing_Heads . That is a magnificent start, but there is no direct mechanism to allow Heads contributors to update it. There is some [additional documentation](https://trmm.net/Heads) on Trammell's website, setting out the Heads rationale and threat model, etc.

As Heads development progresses, it would be useful if external contributors (like me) could update the documentation, not just the code.

@osresearch, would you be willing to migrate that document over to the [Heads repository](https://github.com/osresearch/heads.git) (e.g. in a `/docs` subdirectory), or to the [wiki in the osresearch/heads GitHub project](https://github.com/osresearch/heads/wiki), to make that possible? If you want to be able to vet submissions before they are published within the Heads project, then the former is probably the best option, as it would mean updates to the documentation would come to you as pull requests.

Alternatively, can we request edit rights to the *Installing Heads* page on your website, and update the documentation that way? (I see that your website runs MediaWiki, but the [sign-up page](https://trmm.net/Special:UserLogin/signup) to create a new account is currently blocked, presumably to prevent wiki-spam.)

mfc, 2017-02-26:
agreed. github preferred, can donate whatever is useful from [my documentation](https://github.com/mfc/flashing-docs/blob/master/walkthrough%20for%20flashing%20heads%20on%20an%20x230.md).

osresearch, 2017-02-27:
The github heads.wiki repo sounds like the best plan to me since it integrates well with the rest of the development workflow.  Too bad the two wikis use different formats...

We should also migrate the images; I"m hosting them on flickr (so that trmm.net doesn't have to pay the bandwidth).

osresearch, 2017-03-01:
I imported the page from trmm.net.  Feel free to edit and send pull requests.
https://github.com/osresearch/heads/wiki/Installing-Heads

mfc, 2017-03-01:
FYI you can't do [edits and pull requests for wikis](http://stackoverflow.com/questions/10642928/how-to-pull-request-a-wiki-page-on-github), so I would put this instead as a markdown page in the repo.

osresearch, 2017-03-01:
It seems that there are some benefits to using the wiki:
* Live preview of edits
* Page link syntax
* Multiple wiki language support
* Keeps (unsigned) documentation edits away from (signed) code commits

If the [wiki editing permissions are changed](https://help.github.com/articles/changing-access-permissions-for-wikis/) does it turn into a free-for-all?

mfc, 2017-03-01:
yeah, either create a separate docs repo or always check that docs-related commits are only touching the docs (since you're the only one merging anyway). most of the other features you mention are present in doing markdown edits in a repo (there is a preview button, page link syntax).

osresearch, 2017-03-01:
The page link syntax in github wiki's lets you write `[[Installing Heads]]`, which expands to the equivalent `[Installing Heads](./Installing-Heads.md)`.  This doesn't seem to work in normal markdown pages.

Here's another suggestion -- create a automatically synced repo that pushes to the "official" wiki: http://www.growingwiththeweb.com/2016/07/enabling-pull-requests-on-github-wikis.html

jedahan, 2017-03-01:
That looks like a pretty good workaround. What are people's thoughts on using a separate *wiki* branch on the main repo, as opposed to a separate repo?

osresearch, 2017-03-25:
The github wiki works, but it has some limitations and doesn't seem to let us use a nice URL.  Instead I've cloned the old wiki tree to https://github.com/osresearch/heads-wiki and pointed http://osresearch.net/ at it.  This seems to work, although it has the limitations on syntax that I pointed out last month.

I'll close out the old wiki and set it to point to the new pages, and you can file pull requests or request contributor access to the new pages.

closed 2017-03-25 by osresearch