linuxboot/heads #114: Default build log is so verbose that it prevents Travis CI from working

jonathanfemideer, 2017-02-26:
On 2017-02-05, @blackwellops attempted to [add continuous integration into their Heads fork](https://github.com/blackwellops/heads/commits/master). As far as I can tell, this failed because [the logs were too verbose](https://s3.amazonaws.com/archive.travis-ci.org/jobs/198658402/log.txt):

```
The log length has exceeded the limit of 4 MB (this usually means that the test suite is raising the same exception over and over).

The job has been terminated
```

Should the Heads build be made less verbose by default, to accommodate CI systems with log length limitations like this?

blackwellops, 2017-02-26:
I'm sure something like this could be explored as well. 

https://stackoverflow.com/questions/26082444/how-to-work-around-travis-cis-4mb-output-limit/26082445#26082445

jedahan, 2017-02-27:
also might be worth looking into circle ci, not sure if they have a limit or not

osresearch, 2017-02-28:
Would it make sense to write a per-module make output log?

osresearch, 2017-02-28:
With the log directory, we can see that 7.6 MB of the make output are `musl-cross` and 1.6 MB are `musl`.

osresearch, 2017-02-28:
Although it would be nice to still have the cbfs printout at the end.

closed 2017-03-01 by osresearch

jonathanfemideer, 2017-03-01:
> Would it make sense to write a per-module make output log?

Sounds good.

> Although it would be nice to still have the cbfs printout at the end.

Maybe this could be summarized in the main log?