linuxboot/heads #116: ISO9660 not supported

jonathanfemideer, 2017-02-26:
It would be nice to be able to install Qubes from DVD. The Qubes maintainers note the [advantage of optical media](https://www.qubes-os.org/doc/install-security/): write-resistance. That reduces the risk of installation media being tampered with.

However, the [ISO 9660](https://en.wikipedia.org/wiki/ISO_9660) file system is not supported in Heads. If possible, can support for it be added? If not, then this limitation should at least be documented, to avoid users wasting time burning a Qubes install DVD and trying to install from it using Heads. (Ask me how I know this! :wink:)

osresearch, 2017-02-27:
Perhaps if the ISO9660 driver is built as a module; one of the plans is to have the recovery shell change the PCRs so that any attempts to load external code (such as by mounting a DVD or USB) won't allow unauthorized access to the TPM secrets (issue #16).

osresearch, 2017-03-20:
Fixed! I've added the ISO9660 filesystem.

closed 2017-03-20 by osresearch