linuxboot/heads #117: Can't own/set my TPM for tpmtotp

JohnnyLeone, 2017-02-27:
Hardware: Thinkpad X230

If i try to reset and own the tpm as described [here](https://trmm.net/Installing_Heads#Configuring_the_TPM) i get as response from takeown (currently i'm using takeclear as it seems that there is a naming issue in the tpm command) that says `Warning: srk.pem is not supported`. Every previous command runs without error, also the tpm forceclear.
I've also tried to set the tpm with the stock bios to my preferred SRK and password with success. Using then heads it fails with sealtotp.sh at `Unable to seal secret`.
As far as i tested already (coreboot 4.5 and HEAD) i unable to use the tpm with coreboot on the x230 it appears to be deactivated (Correct tpm flags are set in the coreboot config), i can only set/reset the tpm with the stock bios.

Thanks

mfc, 2017-02-27:
same warning/problem. afterward I can "successfully" run `sealtotp.sh` with the associated password I attempted with `tpm takclear` which gives me a QR code, however running `unsealtotp.sh` the code generated does not match the code generated in my OTP app.

JohnnyLeone, 2017-02-27:
After some more tests with the build in tpm tool in heads it seems to work also with the warning from first post. Sometimes i had a error message after starting `sealtotp.sh`.
Last few attemps it works also with the printed QR code and the verification from my used app on my phone. However after a restart i have to start the process all over again to own the TPM.

mfc, 2017-02-27:
can you boot into Qubes regardless? i'm also on an x230, encountering [this issue](https://github.com/osresearch/heads/issues/119).

osresearch, 2017-02-27:
The `srk.pem is not supported` message is due to a dependency on the (huge) OpenSSL library in `takeown.c`.  This could likely be converted to mbedtls, but didn't seem pressing. https://github.com/osresearch/tpmtotp/blob/master/util/takeown.c#L320

And it does look like a bug in the naming. Whoops. https://github.com/osresearch/tpmtotp/blob/master/util/takeown.c#L361 should be `"takeown"`.

osresearch, 2017-02-27:
@mfc the code probably doesn't match due to a timezone issue. Do you have the hardware clock in local or UTC?

mfc, 2017-02-28:
> the code probably doesn't match due to a timezone issue. Do you have the hardware clock in local or UTC?

`date` and `date -u` shows UTC, with local time same as UTC. I switched to actual local time with `export TZ=TIMEZONE`, and correct timezone is shown for `date` and `hwclock -u`, however `hwclock --localtime` says UTC. The codes still don't match. during boot says `clocksource: Switch to clocksource tsc` and hangs for a long period of time (~5min) for `random: crng init done`.

Anyway this is a different issue. For the issue at hand, even when set (wrongly) it disappears after boot: 

> However after a restart i have to start the process all over again to own the TPM.

it starts with:
```
TPM TOTP:
random: tpm: uninitialized urandom read (20 bytes read)
Error TPM deactivated from TPM_Unseal
Unable to unseal totp secret
```

mfc, 2017-03-02:
holy crap got this to work! it required retrying the commands a bunch of times, rebooting a bunch to fix the state (as [Trammel mentioned](https://github.com/osresearch/heads/wiki/Installing-Heads#taking-ownership)). the codes are a bit out of sync (~60% time overlap) but that's Functional.

osresearch, 2017-03-28:
If you totp codes are out of sync, it is likely a clock issue.  On my test machines ntp doesn't run often enough, so I tend to see several seconds of skew.  Maybe this could be enhanced to compute and display the previous, current and next TOTP code simultaneously.

closed 2017-03-28 by osresearch