linuxboot/heads #119: document adding signing key?

mfc, 2017-02-27:
i can't `start-xen` because it says:

```
gpgv: can't open `/boot/xen-4.6.3.gz.asc`
gpgv: verify signatures failed: file open error
Xen signature failed
```

not sure if this is because I need to first add my pgp key? if so, some basic documentation would be appreciated.

JohnnyLeone, 2017-02-27:
You have to modify the start script `bin/start-xen` and comment those gpg verification lines out. I think those checks are for further implementation of signing the xen binary you're building in the heads tree. As reference you can have a look at this issue #110, there is the process pretty detailed described.
To prevent to change the file every time you boot the system you have to change them in the heads repository and build and flash the image to you're device.

osresearch, 2017-02-27:
@JohnnyLeone is right -- you can comment out the verification until you have added your key and signed your binaries in `/boot`.  We need to document how to do that.

mfc, 2017-03-02:
can you provide a basic rundown of the process? 
1. confirmation that this happens prior to `make` 
2. add my key to `/heads/initrd/.gnupg/trustedkeys.gpg`? (`gpg --no-default-keyring --keyring=/heads/initrd/.gnupg/trustedkeys.gpg --import /my-public-key.asc`)
3. what to sign (and where to put the signed files so that they show up in `/boot`)

mfc, 2017-03-14:
more generally I'm interested in some leads on:
https://github.com/osresearch/heads/wiki/Installing-Heads#read-only-root https://github.com/osresearch/heads/wiki/Installing-Heads#hashing-the--partition-and-setting-up-dm-verity https://github.com/osresearch/heads/wiki/Installing-Heads#signing-boot

if folks know of any existing documentation on it please share.

tlaurion, 2017-06-01:
Signing trough qubes-update script and a gpg compliant card results in the following with hrafs 0.2.1 while following the actual procedure to include keys into the initrd.

Any hint ?

tlaurion, 2017-06-03:
Seems to be linked to to an inconsistency between gpg and gpg in debian8, from which my initrd.cpio got generated. 

Will investigate: https://superuser.com/questions/1112673/gpg2-no-secret-key

tlaurion, 2017-06-14:
Yep. Inconsistencies. My problem was in regard of having moved 4096 subkeys to card and attempting to use them to sign files with it. This is supported only in latest version of gpg (2.1.21).

Note that the current documentation doesn't link to https://github.com/osresearch/heads-wiki/blob/master/GPG.md, which resolves the issue.

Also note that when initrd is rebuilt to include crypttab pointing to /secret.key after which dracut --force is emmited from dom0, the hashes are not valid anymore, which doesn't release /secret.key and results in PCR #4 error. The fix is to actually delete the /boot/boot.hashes* and reboot to heads. A qubes-update from there fixes it all.

Will redo each steps and propose PR in the next couples of day in documentation to that it reflects required steps.

Good job Trammel and al.!

tlaurion, 2019-05-21:
Fixed with GPG2

tlaurion, 2019-05-21:
Note: proper commands in https://github.com/osresearch/heads-wiki/blob/master/Installing-Heads.md

[Commands were added into ash history](https://github.com/osresearch/heads/blob/master/initrd/.ash_history#L8), so scrolling up in recovery shell history proposes command templates

Using Whiptail permits to deal with this graphically now, and his the default board configuration for X230. More old boards and newly integrated ones are also defaulting to FBWhiptail now.

@mfc, tag me in if it is still confusing.

closed 2019-05-21 by tlaurion