linuxboot/heads #120: SHA-1 hash collision - implications for Heads?

jonathanfemideer, 2017-02-27:
Heads relies on the X230's TPM 1.2, which [relies on SHA-1 in certain ways](https://trustedcomputinggroup.org/sha-1-uses-tpm-v1-2/).

Heads also relies on GnuPG, which also [relies on SHA-1](https://lists.gnupg.org/pipermail/gnupg-users/2017-February/057789.html) in [some ways](http://www.sami-lehtinen.net/blog/sha1-sha-1-is-broken-how-to-configure-gnupg-and-mozilla-thunderbird-enigmail).

SHA-1 was recently [broken](https://shattered.it). What are the implications for Heads?

Do any parts of Heads need to be re-written or re-thought to mitigate the impact, or does Heads only use SHA-1 in ways that sidestep the collision concern?

osresearch, 2017-02-27:
We're using `sha256sum` for the various dependencies, so hopefully that will be secure for a while.

Since the TPM PCR extend operation isn't subject to chosen plaintext pre-image attacks, it is unlikely that the TPM can be subverted with the new attack.  The problem is that the TPM hardware defines the standard; are there any machines with discrete TPM 2.0 chips? The ME based fTPM has its own set of concerns.

The GPG signatures are a potential concern. I believe we can work around it by specifying the appropriate hash algorithm with `gpg --digest-algo SHA256`.  If we want to improve things, we could patch `gpgv` to not allow sha-1 signatures.

osresearch, 2017-05-04:
The `qubes-update` signing script now explicitly uses SHA256 to avoid collisions.

Since we control what is measured and *how* it is measured, we could modify the TPM extension code to perform a SHA256 of the memory or content and then let the TPM extend the PCR with the SHA1 of the SHA256 hash.

closed 2017-05-04 by osresearch

jpouellet, 2017-06-02:
:+1: 

It appears not unreasonable to assume that [well-resourced adversaries can collide SHA-1 at will by now](https://theintercept.com/2017/05/11/nyu-accidentally-exposed-military-code-breaking-computer-project-to-entire-internet/).

I am not a cryptographer, but my understanding is that colliding SHA-1 on feasible timelines is likely to require the introduction of carefully chosen disturbance vectors in an input block(s), and a SHA1(SHA256(thing_to_be_measured)) construction appears intuitively to mitigate the ability for adversaries to introduce such disturbance vectors.

From [*The first collision for full SHA-1*](http://shattered.it/static/shattered.pdf) (emphasis mine):
> To ensure all message word bit differences are compatible with the linear message
> expansion, one uses a Disturbance Vector (DV) [4] that is a correctly expanded message
> itself, but where every “1” bit marks the start of a local collision. **The selection of a good
> disturbance vector is very important for the overall attack cost.** As previously shown by
> Wang et al. [42], the main reason of using two block pairs instead of just one pair is that
> this choice alleviates an important restriction on the disturbance vector, namely that there
> are no state differences after the last step. Similarly, it may be impossible to unite the
> input chaining value difference with the local collisions for an arbitrary disturbance vector.

However, again, "I am not a cryptographer, and this is not legal advice" ;)