linuxboot/heads #121: License is open to Tivoization

jonathanfemideer, 2017-03-01:
Thanks for [licensing Heads under the GPLv2](https://github.com/osresearch/heads/commit/99990ab7153377174ea0fdd6fecc3746064bfdf0) and closing #115. GPLv2 is definitely better than nothing!

However, it leaves the door open to Heads being used in a device in a manner that the device's owner is powerless to affect: exactly the kind of thing Heads seems intended to prevent. (I'm talking about [Tivoization](https://en.wikipedia.org/wiki/Tivoization)). Please could you at least bump the license up to GPLv3 to prevent that?

jonathanfemideer, 2017-03-02:
@jpouellet, I see you have thumbed this down. Can you provide a use-case where it is in the device owner's interest for Tivoization to be allowed by the license?

jpouellet, 2017-03-03:
No, but I have gotten quite tired of re-implementing things and duplicating perfectly good effort due to wanting to bring gpl-restricted code to more-permissive (BSD-style or GPL2-only) projects whose maintainers do not wish to add additional restrictions to their own code by introducing code with infectious licenses such as the GPL.

Since the status quo is boot sequences being entirely proprietary, I think it makes sense for this effort to be as widely usable as possible to maximize its potential impact.

That said, I have contributed no effort to this project, so my opinion shouldn't carry much (if any) weight. It was more of a "sigh..." response than a "thou shalt not do ____".

jonathanfemideer, 2017-03-03:
@jpouellet 
> the status quo is boot sequences being entirely proprietary and insecure

Yes, but permissive licensing strengthens the hands of those who create proprietary software. And once incorporated into a proprietary or Tivoized boot sequence, it would be much harder to tell whether such a Heads implementation were secure.

The only way to legally prevent this would be to adopt a license that would do so.

>  I have gotten quite tired of re-implementing things and duplicating perfectly good effort due to wanting to bring gpl-restricted code to more-permissive (BSD-style or GPL2-only) projects

I sympathize with your desire to see other projects able to benefit from Heads. I do not think they would be prevented from doing so by Heads adopting GPLv3 or AGPLv3, because Heads can and almost certainly would be built separately. Suppose you wanted to use Heads to boot FreeBSD: as long as your FreeBSD install were capable of being booted from Heads, it would work. The work to suitably modify FreeBSD in that way needn't be done as part of Heads: it could (and arguably should) be done as part of the FreeBSD project. Indeed, FreeBSD users have long been asking questions about making FreeBSD kexec-able or similar, and the GPL does not seem to have been the obstacle:

* https://wiki.freebsd.org/Kload
* https://forums.freebsd.org/threads/59123/
* https://lists.freebsd.org/pipermail/freebsd-hackers/2011-November/036820.html

tlaurion, 2020-03-09:
@osresearch This is important.

closed 2024-04-16 by tlaurion