linuxboot/heads #123: [UX] Improve user experience

mfc, 2017-03-02:
currently, to boot your operating system (Qubes) with Heads:

1. turn on machine
2. read & confirm OTP code
3. type `start-xen` (not [`./start-xen`](https://github.com/osresearch/heads/issues/122))
4. type `kexec -e`

instead, the process should be:

1. turn on machine
2. read & confirm OTP code
3. press Enter

with dropping into the prompt requiring pressing ESC or similar.

osresearch, 2017-03-02:
Possible reasons to drop to the shell:

* No `/boot` partition
* No `/boot/boot.sh` script
* Failed signatures
* `kexec` failure

Things that we need to streamline:

* TPM counter based version checking
* TPM keys and dom0 initrd
* PCR values for known good builds
* PCR updates after reading the TOTP and before starting Xen

osresearch, 2017-03-31:
Reopening since we haven't actually implemented the startup scripts for Qubes in the new configurable build.  All the pieces are there, just need to make them happen.

osresearch, 2017-04-03:
The startup is almost seamless, although quite verbose.  The last remaining issue is #29 which might need a systemd fix unless we figure out how to properly generate the `/etc/crypttab` file in the Qubes' initrd image.  Currently it does not have all of the parititions listed.

osresearch, 2017-04-12:
On a configured and sealed system the startup is greatly improved.  It is possible to boot all the way into Qubes' login screen with just the disk unlock code.

closed 2017-04-12 by osresearch