linuxboot/heads #126: QEMU image reboots after loading payload

paulmenzel, 2017-03-03:
Using commit 45324e83c2 (remove converted wiki pages for now), running `make BOARD=qemu`, and starting that with the command below, it doesn’t boot.

```
$ dpkg -l qemu-system-x86
[…]
ii  qemu-system-x86                                1:2.6.1+dfsg-0ubuntu5.3      amd64                        QEMU full system emulation binaries (x86)
$ cbfstool ~/heads-qemu-image.rom print
Name                           Offset     Type         Size
cbfs master header             0x0        cbfs header  32
fallback/romstage              0x80       stage        14524
cpu_microcode_blob.bin         0x39c0     microcode    0
fallback/ramstage              0x3a40     stage        49058
fallback/dsdt.aml              0xfa40     raw          6952
fallback/payload               0x115c0    payload      2792608
(empty)                        0x2bb2c0   null         5523544
bootblock                      0x7ffb40   bootblock    848
$ qemu-system-i386 -machine q35 -chardev file,id=debugcon,path=/dev/shm/debugcon.log -device isa-debugcon,iobase=0x402,chardev=debugcon -serial stdio -bios ~/heads-qemu-image.rom | tee /dev/shm/heads.txt
[…]
IMD small region:
  IMD ROOT    0. 07ffec00 00000400
  CAR GLOBALS 1. 07ffeac0 00000140
  ROMSTAGE    2. 07ffeaa0 00000004
  GDT         3. 07ffe8a0 00000200
CBFS: 'Master Header Locator' located CBFS at [400100:bfffc0)
CBFS: Locating 'fallback/payload'
CBFS: Found @ offset 115c0 size 2a9ca0
Loading segment from ROM address 0xff8116f8
  data (compression=0)
  New segment dstaddr 0x90000 memsize 0x1080 srcaddr 0xff811784 filesize 0x1080
Loading segment from ROM address 0xff811714
  code (compression=0)
  New segment dstaddr 0x1000000 memsize 0x2a8ac0 srcaddr 0xff812804 filesize 0x2a8ac0
Loading segment from ROM address 0xff811730
  code (compression=0)
  New segment dstaddr 0x40000 memsize 0xb1 srcaddr 0xffabb2c4 filesize 0xb1
Loading segment from ROM address 0xff81174c
  data (compression=0)
  New segment dstaddr 0x91000 memsize 0x23 srcaddr 0xffabb375 filesize 0x23
Loading segment from ROM address 0xff811768
  Entry Point 0x00040000
Bounce Buffer at 07f5a000, 354528 bytes
Loading Segment: addr: 0x0000000000090000 memsz: 0x0000000000001080 filesz: 0x0000000000001080
lb: [0x0000000000100000, 0x000000000012b470)
Post relocation: addr: 0x0000000000090000 memsz: 0x0000000000001080 filesz: 0x0000000000001080
it's not compressed!
[ 0x00090000, 00091080, 0x00091080) <- ff811784
dest 00090000, end 00091080, bouncebuffer 7f5a000
Loading Segment: addr: 0x0000000001000000 memsz: 0x00000000002a8ac0 filesz: 0x00000000002a8ac0
lb: [0x0000000000100000, 0x000000000012b470)
Post relocation: addr: 0x0000000001000000 memsz: 0x00000000002a8ac0 filesz: 0x00000000002a8ac0
it's not compressed!
[ 0x01000000, 012a8ac0, 0x012a8ac0) <- ff812804
dest 01000000, end 012a8ac0, bouncebuffer 7f5a000
Loading Segment: addr: 0x0000000000040000 memsz: 0x00000000000000b1 filesz: 0x00000000000000b1
lb: [0x0000000000100000, 0x000000000012b470)
Post relocation: addr: 0x0000000000040000 memsz: 0x00000000000000b1 filesz: 0x00000000000000b1
it's not compressed!
[ 0x00040000, 000400b1, 0x000400b1) <- ffabb2c4
dest 00040000, end 000400b1, bouncebuffer 7f5a000
Loading Segment: addr: 0x0000000000091000 memsz: 0x0000000000000023 filesz: 0x0000000000000023
lb: [0x0000000000100000, 0x000000000012b470)
Post relocation: addr: 0x0000000000091000 memsz: 0x0000000000000023 filesz: 0x0000000000000023
it's not compressed!
[ 0x00091000, 00091023, 0x00091023) <- ffabb375
dest 00091000, end 00091023, bouncebuffer 7f5a000
Loaded segments
ICH7 watchdog disabled
Jumping to boot code at 00040000(07fd6000)
CPU0: stack: 00122000 - 00123000, lowest used address 00122c30, stack used: 976 bytes
entry    = 0x00040000
lb_start = 0x00100000
lb_size  = 0x0002b470
buffer   = 0x07f5a000
QEMU debugcon detected [port 0x402]

coreboot-TIMELESS--heads Thu Jan  1 00:00:00 UTC 1970 romstage starting...
CBMEM:
qemu: fatal: Trying to execute code outside RAM or ROM at 0x692f3533

EAX=00000000 EBX=00000000 ECX=00000053 EDX=00000000
ESI=00000000 EDI=ff803811 EBP=fffffffe ESP=ffffff6c
EIP=692f3533 EFL=00000012 [----A--] CPL=0 II=0 A20=1 SMM=0 HLT=0
ES =0010 00000000 ffffffff 00cf9300 DPL=0 DS   [-WA]
CS =0008 00000000 ffffffff 00cf9b00 DPL=0 CS32 [-RA]
SS =0010 00000000 ffffffff 00cf9300 DPL=0 DS   [-WA]
DS =0010 00000000 ffffffff 00cf9300 DPL=0 DS   [-WA]
FS =0010 00000000 ffffffff 00cf9300 DPL=0 DS   [-WA]
GS =0010 00000000 ffffffff 00cf9300 DPL=0 DS   [-WA]
LDT=0000 00000000 0000ffff 00008200 DPL=0 LDT
TR =0000 00000000 0000ffff 00008b00 DPL=0 TSS32-busy
GDT=     ff800200 0000001f
IDT=     00000000 00000000
CR0=60000011 CR2=00000000 CR3=00000000 CR4=00000000
DR0=00000000 DR1=00000000 DR2=00000000 DR3=00000000
DR6=ffff0ff0 DR7=00000400
CCS=00000008 CCD=0039685b CCO=SUBL
EFER=0000000000000000
FCW=037f FSW=0000 [ST=0] FTW=00 MXCSR=00001f80
FPR0=0000000000000000 0000 FPR1=0000000000000000 0000
FPR2=0000000000000000 0000 FPR3=0000000000000000 0000
FPR4=0000000000000000 0000 FPR5=0000000000000000 0000
FPR6=0000000000000000 0000 FPR7=0000000000000000 0000
XMM00=00000000000000000000000000000000 XMM01=00000000000000000000000000000000
XMM02=00000000000000000000000000000000 XMM03=00000000000000000000000000000000
XMM04=00000000000000000000000000000000 XMM05=00000000000000000000000000000000
XMM06=00000000000000000000000000000000 XMM07=00000000000000000000000000000000
```

So it somehow restarts in the end.

Please find the full log attached.
[heads.txt](https://github.com/osresearch/heads/files/817213/heads.txt)

osresearch, 2017-03-03:
I'm running an older version, but that might not be the problem:

```
% qemu-system-x86_64 --version
QEMU emulator version 2.3.0 (Debian 1:2.3+dfsg-5ubuntu9.4), Copyright (c) 2003-2008 Fabrice Bellard
```

I note that your payload region is quite small -- the last reproducible build is 3.3 MB, yours is only 2.8 MB.  Have you changed something in the initrd or kernel image?

```
./build/coreboot-4.5/qemu/cbfstool qemu.rom print
Performing operation on 'COREBOOT' region...
Name                           Offset     Type         Size
cbfs master header             0x0        cbfs header  32
fallback/romstage              0x80       stage        14588
cpu_microcode_blob.bin         0x3a00     microcode    0
fallback/ramstage              0x3a80     stage        49092
fallback/dsdt.aml              0xfa80     raw          6952
fallback/payload               0x11600    payload      3307040
(empty)                        0x338c80   null         5009048
bootblock                      0x7ffb40   bootblock    848
```

paulmenzel, 2017-03-03:
Did you use also latest master version?

To my knowledge I haven’t changed anything. I guess our compilers differ for building the Linux kernel?

The file `bzImage` is 2803904 kB (2.7 MB) big.

osresearch, 2017-03-03:
If you're getting a different kernel image, then our reproducible builds have failed... Is there an initrd in that kernel? (Although that would make a 1.2 MB difference, not a 0.5 MB change).

paulmenzel, 2017-03-03:
How can I check that? (I’ll can find that out on Monday.)

osresearch, 2017-03-03:
You should have a `heads/initrd.cpio` file in the top level directory, and it should be about 3.3 MB.
```
diamond:~/clean/heads: ls -Fla initrd.cpio
-rw-rw-r-- 1 hudson hudson 3097664 Mar  3 11:56 initrd.cpio
diamond:~/clean/heads: sha256sum initrd.cpio
8228a7b00348eb43176c38f5f61e4d0fa08584fbd1d7a9ad04e0dc1fecf4e9a8  initrd.cpio
```

The files that go into it should have the following hashes:

```
diamond:~/clean/heads: find initrd -type f -print0 | xargs -0 sha256sum | sort
0169d966aaabb5f12709c0d831c096dce7d2bb471bd959fab969592023e1d090  initrd/bin/wrap-cpio
168bd27c33645acd64c55673aff7baf314c05960758d32448bd44b9b9575912c  initrd/bin/busybox
271891a97cfabcc94f4bbe6823d7277279080583a07bb214f9bdbf4dcb1fbc47  initrd/bin/seal-key
298041c3e7e36eb8ff7385c99fa1cb1ec3d1cd59fb66fba56c540525f6ead28e  initrd/bin/generate-crypttab
357bdfa471482578709671dd7103c0dd56bd0c418e1c5eb8f2ad0861e6a19993  initrd/bin/cbmem
35acd9f6c0aec36bb19aae065302135e41656b9858c6d8301666b951d259b587  initrd/bin/start-xen
3cb885da20eac8413d78fbcf0e941b64e4d13b76d271b271ffcca709e5ec6d83  initrd/bin/gpgv
438999b9091448b371af7242e898fe0c0bb58dc228d8dd53793aadc19db8144b  initrd/bin/kexec
44c4b4d8604b06f833af931a2374eb60255c25209118a84de6bc4dc62722f422  initrd/lib/libcryptsetup.so.4
5518948d3ff952f970aa4b2ed79f0bf2ef72365b1fce5cb7d4ec83bc0cddc301  initrd/lib/libtpm.so
5d7ad25a0174fdce4db62933897dc33ecfd3421b29c54720d1863a60dcb8adaa  initrd/bin/unsealtotp.sh
5e819f4e4e8fbbde4ef0beaaa4d54100676544b7aa6b1d190bacdf93e757fe15  initrd/bin/sealtotp.sh
60d48fe6acd2884794f7fea5965b16a6a5d82f85b45af0dd4712c8faa80c4ba1  initrd/bin/totp
62a87d9d1dc535a07a2fcbaf9f7111b5bd5bea39789d982be8bcdc9fdec2f37d  initrd/lib/libpopt.so.0
6bea2932d6d9b09aaea6bb11d27a1318b7ba9ebd3fb87b479c21df9de8dc62fb  initrd/bin/dmsetup
73e5156ee067ed725d8cfba1e30bb497a83710d584afff0a88f1589e64d43252  initrd/bin/cryptsetup
7b6e9520c5896fe9eb3269d00d39b0d0654d547a2fbb7eb0402b4ff8fc121422  initrd/lib/libuuid.so.1
7cd945c245d7d9a491c555ef5251a7e70bb08b3b1179f02e1c25e2e726fad2a8  initrd/bin/qrenc
876923a1f71eb5d9f2ce2d385e9e1320b4c3f7aaa35e5fca193f88fa5f42cd92  initrd/bin/base32
895773893602c8a420511d4ded66a53b2d230e8ff23fd66436ad9a30482e8686  initrd/lib/libqrencode.so.3
8e7bafcfe9bf026d2b427717cd734ea8f386d60a4aa3da224f3fd3527e9cddfa  initrd/lib/libmbedcrypto.so.0
9110957ffbaeb3d5732aa3383064d41f3f70418e2fc0d1c8a2f02125595e35b7  initrd/bin/tpm
97fb9760c152b06b9875c60d2a81791e6ec6eabf671b819f0ddcc1393bafc296  initrd/mount-boot
9d04da7b68590eff6f332f2f4cfe5e02a68f7d5e82dfc1329f7f93c4319d0cd4  initrd/bin/unseal-key
adcdf1b41ec9361d57e9c507de5afbd60dada45e2fff60c44c05f009b82ab185  initrd/bin/veritysetup
af33f2d31616acf2475650811e03443cd92b07c4b75566d54b303541b3aa5852  initrd/init
b73cd9f8e51aacd885018656e8bb199ca1e45a6ccf406d3883172e8275b0a0ea  initrd/lib/libc.so
c2d908aa011f7e7288c652ef55ae47929c3ab5e0a5fc650592e2517c3ec10fda  initrd/.ash_history
e2ddb068d79609377f86aa5d67e7d89a48e8464efc8c23f10d7d83aa7b99eefb  initrd/.gnupg/trustedkeys.gpg
f09355eff1d0420a3ca4f325b6790564cf594750ab10e98fb34627e1e4657b3e  initrd/lib/libdevmapper.so.1.02
```

paulmenzel, 2017-03-06:
Thank you.

```
$ find initrd -type f -print0 | xargs -0 sha256sum | sort
0169d966aaabb5f12709c0d831c096dce7d2bb471bd959fab969592023e1d090  initrd/bin/wrap-cpio
168bd27c33645acd64c55673aff7baf314c05960758d32448bd44b9b9575912c  initrd/bin/busybox
271891a97cfabcc94f4bbe6823d7277279080583a07bb214f9bdbf4dcb1fbc47  initrd/bin/seal-key
298041c3e7e36eb8ff7385c99fa1cb1ec3d1cd59fb66fba56c540525f6ead28e  initrd/bin/generate-crypttab
357bdfa471482578709671dd7103c0dd56bd0c418e1c5eb8f2ad0861e6a19993  initrd/bin/cbmem
35acd9f6c0aec36bb19aae065302135e41656b9858c6d8301666b951d259b587  initrd/bin/start-xen
3cb885da20eac8413d78fbcf0e941b64e4d13b76d271b271ffcca709e5ec6d83  initrd/bin/gpgv
438999b9091448b371af7242e898fe0c0bb58dc228d8dd53793aadc19db8144b  initrd/bin/kexec
5518948d3ff952f970aa4b2ed79f0bf2ef72365b1fce5cb7d4ec83bc0cddc301  initrd/lib/libtpm.so
5d7ad25a0174fdce4db62933897dc33ecfd3421b29c54720d1863a60dcb8adaa  initrd/bin/unsealtotp.sh
5e819f4e4e8fbbde4ef0beaaa4d54100676544b7aa6b1d190bacdf93e757fe15  initrd/bin/sealtotp.sh
60d48fe6acd2884794f7fea5965b16a6a5d82f85b45af0dd4712c8faa80c4ba1  initrd/bin/totp
6bea2932d6d9b09aaea6bb11d27a1318b7ba9ebd3fb87b479c21df9de8dc62fb  initrd/bin/dmsetup
73e5156ee067ed725d8cfba1e30bb497a83710d584afff0a88f1589e64d43252  initrd/bin/cryptsetup
7cd945c245d7d9a491c555ef5251a7e70bb08b3b1179f02e1c25e2e726fad2a8  initrd/bin/qrenc
876923a1f71eb5d9f2ce2d385e9e1320b4c3f7aaa35e5fca193f88fa5f42cd92  initrd/bin/base32
8e7bafcfe9bf026d2b427717cd734ea8f386d60a4aa3da224f3fd3527e9cddfa  initrd/lib/libmbedcrypto.so.0
9110957ffbaeb3d5732aa3383064d41f3f70418e2fc0d1c8a2f02125595e35b7  initrd/bin/tpm
97fb9760c152b06b9875c60d2a81791e6ec6eabf671b819f0ddcc1393bafc296  initrd/mount-boot
9d04da7b68590eff6f332f2f4cfe5e02a68f7d5e82dfc1329f7f93c4319d0cd4  initrd/bin/unseal-key
adcdf1b41ec9361d57e9c507de5afbd60dada45e2fff60c44c05f009b82ab185  initrd/bin/veritysetup
af33f2d31616acf2475650811e03443cd92b07c4b75566d54b303541b3aa5852  initrd/init
c2d908aa011f7e7288c652ef55ae47929c3ab5e0a5fc650592e2517c3ec10fda  initrd/.ash_history
e2ddb068d79609377f86aa5d67e7d89a48e8464efc8c23f10d7d83aa7b99eefb  initrd/.gnupg/trustedkeys.gpg
```

So the libraries are missing in my image. The hashes of the added files are correct though, that means, are the same as the hashes from your paste.

teaso, 2017-03-08:
Try running with qemu-system-x86_64 rather than qemu-system-i386. Not sure why your initrd is different though.

osresearch, 2017-03-09:
The `populate-libs` script must be failing on your system for some reason. Now that the builds are fully reproducible and don't depend on the system libraries, I suspect that we can discard the script and replace it with something that just copies targets from `./install/lib` rather than using `ldd` to find the libraries.

osresearch, 2017-03-30:
`populate-libs` is no longer a thing. This should be resolved by https://github.com/osresearch/heads/commit/ab0476ad2f20d391c71000770ec39f772a24480d

closed 2017-03-30 by osresearch