linuxboot/heads #134: Server port to ASUS KGPE-D16?

tlaurion, 2017-03-21:
Would that make sense? Using Qubes as a server spinner for Qubes seems like a nice and practical experiment.

1. Having a server that would validate its status prior to passphrase validation (local or even remote [openbmc](https://www.raptorengineering.com/coreboot/kgpe-d16-status.php)).
1. It is possible to use Qubes in [server mode](https://github.com/Rudd-O/qubes-network-server) (kinda)
1. It is possible to have 16mb of flash on that beast, that can boot libreboot/coreboot without binary blobs.

osresearch, 2017-03-21:
This is a very good idea. There is the beginning of a server build of heads in the [moc branch](https://github.com/osresearch/heads/tree/moc) to support the [Mass Open Cloud](https://info.massopencloud.org/) project's end-to-end attested design.

osresearch, 2017-03-21:
Does it have a TPM?  That is a fairly hard requirement for measured boot and the attestation system.

tlaurion, 2017-03-21:
> Does it have a TPM? That is a fairly hard requirement for measured boot and the attestation system.
TPM 	[YES](https://www.coreboot.org/Board:asus/kgpe-d16#Features) 	With Owner Controlled CRTM - TPM is an option addon module

tlaurion, 2017-03-22:
This module 
https://www.amazon.ca/Accessory-FW3-19-Module-BitLocker-Retail/dp/B007V9RQLY

tlaurion, 2017-05-08:
Supports Infineon module.
http://anzwix.com/a/Coreboot/Mbasuskgped16AddTPMSupport

tlaurion, 2017-11-20:
Forgot to update : OpenBMC port was made. Need to test. 
https://www.raptorengineering.com/coreboot/kgpe-d16-bmc-port-status.php

tlaurion, 2018-01-28:
@flammit @osresearch 

Porting is on the [way](https://github.com/tlaurion/heads/tree/KGPE-D16_PORT-BOOT-NoTpm)! 
Heads boots, but the TPM is wrongly detected as a version 1.2 instead of [2.0](https://ecc2017.coreboot.org/uploads/talk/presentation/37/Enabling_TPM2.pdf) resulting in ownership being impossible. Investigating.
The branch includes RaptorEngineer's flashrom patches so that OpenBMC can be flashed from within Heads.

Actual boot trace: [KGPE-D16-No_TPM.txt](https://github.com/osresearch/heads/files/1670790/KGPE-D16-No_TPM.txt)

tlaurion, 2018-01-28:
@zaolin : Any input on TPM initialization issues encountered?

osresearch, 2018-01-28:
This is great news.  Glad to hear that progress is being made on the KGPE-D16.

Regarding TPM 2.0, we really need to figure out the best way to work with them. Issue #287 begins to address it, but we'll also need to find a lightweight library that works well with it (and decide if we want to support fTPM systems).

paulmenzel, 2018-01-28:
@tlaurion, please bring the issue up on the coreboot mailing list or submit an issue to [the coreboot bug tracker](https://ticket.coreboot.org) with your config, the commit hash and the full boot log.

paulmenzel, 2018-01-28:
Also, as far as I understood the Linux TPM folks, the Linux kernel should be able to work around broken firmware and set up the TPM by itself. If not, please try with latest Linux master, and report an issue to the Linux kernel TPM folks.

tlaurion, 2018-01-31:
@paulmenzel: the thing is that we want TPM being initialised and used to measure romstage and ramstage BEFORE Linux enters fully in play.

This [patch](https://review.coreboot.org/#/c/coreboot/+/21983/) resolves the issue of SLB9665 TPM2.0 being detected and initialized as a SLB9660 TPM1.2. 

Here is my [patch](https://github.com/tlaurion/heads/commits/KGPE-D16_PORT-BOOT-NoTpm) against flammit's heads' coreboot 4.7, and my [branch](https://github.com/tlaurion/heads/tree/KGPE-D16_PORT-BOOT-NoTpm), missing proper TPM2.0 tools to use the TPM.

Boot log exerpt:
> lpc_tpm: Read reg 0xf00 returns 0x1a15d1
> Found TPM SLB9665 TT 2.0 by Infineon
> TPM: Open

paulmenzel, 2018-01-31:
On 01/31/18 17:40, tlaurion wrote:
> @paulmenzel: the thing is that we want TPM being initialised and used to measure romstage and ramstage BEFORE Linux enters fully in play.

@tlaurion, sorry for the misunderstanding. I am aware of that. Checking 
if Linux is able to correctly set it up was just to make sure that it is 
working in the first place.

[…]

tlaurion, 2018-02-23:
Work continues [here](https://github.com/tlaurion/heads/tree/KGPE-D16_PORT-BOOT-NoTpm).

Heads boots, initializes SL9665 TPM 2.0 chipset, but usage of TPM fails since libraries do not implement 2.0.
Patch set includes raptorengineering's flashrom to flash openbmc from within heads (tested: works), contains mods so that kvm chip can obtain memory access prio to releasing it to coreboot so openbmc works, permits internal flashing (updates) of kgpe-d16 roms from within heads.

Missing: 

- [TPM2.0 toolset](https://github.com/osresearch/heads/issues/287) to properly acquire, measure and attest system integrity.
- Working console to have heads attestation over OpenBMC once TPM2 is implemented.
- Working Xen boot console parameters to interface properly with OpenBMC, so that it would be possible to jump from SSH connection to OpenBMC directly into dom0.
---
make CONFIG=config/kgpe-d16-generic.config -j4
sha256sum kgpe-d16.rom 
0d1f6f20b12026b7f20a33a1930957c9e3ecc539384401426e88986a5d32bab3  kgpe-d16.rom

tlaurion, 2018-02-27:
Update.

Flammit modifications:
- Ported to the new build system. Building the kgpe-d16 image is made by make `BOARD=kgpe-d16`
- Reduced the coreboot patches needed to the minimum, permitting kvm chip to boot.

-Usage:
- System boots without outputting console to the graphic card until the kernel setups it's console configuration. Connect to the kvm chip which gets it's IP from DHCP. From there, connect to the system serial console with `screen /dev/ttyS0 115200`.
- Flashing heads update firmwares is done through `flashrom-kgpe-d16.sh /media/coreboot.rom`
- Qubes disk password is asked through OpenBMC console.

Limitation:
- It is currently impossible to reflash the kvm chip from within the kvm. It's possible to do it from the console connection. I do it from a serial to usb connection from my Qubes laptop through `sudo minicom  -D /dev/ttyUSB0` then `flashrom-kgpe-d16-openbmc.sh /media/flash-asus-20180122172732`. Do not forget to turn off REST api access and changing default ssh password when [building OpenBMC](https://www.raptorengineering.com/coreboot/kgpe-d16-bmc-port-status.php)

tlaurion, 2018-03-02:
@osresearch
https://github.com/osresearch/heads/pull/335

tlaurion, 2018-03-10:
KGPE-D16 support is merged to Heads!

Todo:
- Incorporate latest patches for coreboot to properly support KVM ([more proper integration](https://review.coreboot.org/#/c/coreboot/+/19820/1))
- Port for VBOOT support so that [trusted boot is supported with multiple firmware copies for validation](https://www.youtube.com/watch?v=4EvTcfcYfMY&amp;t) (we have 16mb of flash so no space constraints.) Patchset is available [here](https://review.coreboot.org/#/c/coreboot/+/23756)

closed 2018-03-16 by osresearch