linuxboot/heads #139: Merge MOC and master Linux config, measure loaded modules

osresearch, 2017-03-28:
The main differences are the use of the networking drivers.  These should be build as modules and measured/loadable at runtime rather than requiring compile time decisions.

osresearch, 2017-03-28:
Modules are huge and blow out our space budget.  How do we strip them?

osresearch, 2017-03-28:
`strip --strip-debug` seems to be safe.

rofl0r, 2017-03-28:
did you consider building the modules statically into the kernel ? i.e. `=y` instead of `=m`.
we do that in sabotage and it simplifies the startup and increases robustness.

osresearch, 2017-03-28:
Right now they are statically compiled into the kernel, but for security reasons we want to make the modules loadable.  This reduces the attack surface if the network device is not brought online during boot, or if there is no way to attach a USB storage device during the boot process.  Having the modules loadable, but measured, makes it possible to be flexible during setup and recovery stages, while not also having those boundaries exposed.

rofl0r, 2017-03-28:
good point, that makes sense.

osresearch, 2017-03-30:
Everything is a module now, I think we're clear to merge the branches.  The MOC branch will continue to iterate on their `initrd/init` script.

closed 2017-03-30 by osresearch

osresearch, 2017-03-30:
https://github.com/osresearch/heads/commit/8343130e9ae3b02045cd430664eb2db06d104de1