linuxboot/heads #145: `build.sh` fails with `crossgcc/x86_64-linux-musl/bin: No such file or directory`

paulmenzel, 2017-03-31:
```
$ git describe --tags
v0.1.0-72-gd8ab8ec
$ git log --oneline -1
d8ab8ec Rebuild sub-modules on each build (issue #143).
```

`make BOARD=qemu` fails with the error below.

```
[…]
Location: https://codeload.github.com/sabotage-linux/kernel-headers/tar.gz/v3.12.6-5 [following]
--2017-03-31 12:02:31--  https://codeload.github.com/sabotage-linux/kernel-headers/tar.gz/v3.12.6-5
Resolving codeload.github.com... 192.30.253.121, 192.30.253.120
Connecting to codeload.github.com|192.30.253.121|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: ‘/dev/shm/heads/packages/kernel-headers-3.12.6-5.tar.gz’

/dev/shm/heads/packages/kernel-headers-3.12.6-5.tar.     [   <=>                                                                                                                ] 988.47K  1.58MB/s    in 0.6s

2017-03-31 12:02:32 (1.58 MB/s) - ‘/dev/shm/heads/packages/kernel-headers-3.12.6-5.tar.gz’ saved [1012195]

echo "ecf4db8781dc50a21cbc4cb17b039f96aede53f9da13435a3201373abb49b96b  /dev/shm/heads/packages/kernel-headers-3.12.6-5.tar.gz" | sha256sum --check -
/dev/shm/heads/packages/kernel-headers-3.12.6-5.tar.gz: OK
tar -xf "/dev/shm/heads/packages/kernel-headers-3.12.6-5.tar.gz" -C "/dev/shm/heads/build"
if [ -r patches/kernel-headers-3.12.6-5.patch ]; then ( cd /dev/shm/heads/build/kernel-headers-3.12.6-5 ; patch -p1 ) < patches/kernel-headers-3.12.6-5.patch; fi
2017-03-31 12:02:32+02:00 CONFIG kernel-headers
2017-03-31 12:02:32+02:00 MAKE kernel-headers
2017-03-31 12:02:42+02:00 MAKE musl-cross
tail /dev/shm/heads/build/log/musl-cross.log
-----
+ build 2 gcc-5.3.0 --target=x86_64-linux-musl --enable-languages=c,c++ --disable-libmudflap --disable-libsanitizer --disable-nls --disable-multilib --with-multilib-list= --disable-shared
+ BP=2
+ BD=gcc-5.3.0
+ CF=./configure
+ BUILT=/dev/shm/heads/build/musl-cross-git/gcc-5.3.0/built2
+ shift
+ shift
+ '[' '!' -e /dev/shm/heads/build/musl-cross-git/gcc-5.3.0/built2 ']'
+ doinstall 2 gcc-5.3.0
+ BP=2
+ BD=gcc-5.3.0
+ INSTALLED=/dev/shm/heads/build/musl-cross-git/gcc-5.3.0/installed2
+ shift
+ shift
+ '[' '!' -e /dev/shm/heads/build/musl-cross-git/gcc-5.3.0/installed2 ']'
+ rm -rf '/dev/shm/heads/crossgcc/x86_64-linux-musl/lib/gcc/x86_64-linux-musl/*/include-fixed/' '/dev/shm/heads/crossgcc/x86_64-linux-musl/lib/gcc/x86_64-linux-musl/*/include/stddef.h'
+ cd /dev/shm/heads/crossgcc/x86_64-linux-musl/bin
./build.sh: line 147: cd: /dev/shm/heads/crossgcc/x86_64-linux-musl/bin: No such file or directory
make[2]: *** [Makefile:2: ../../crossgcc/x86_64-linux-musl/bin/x86_64-linux-musl-gcc] Error 1
make[2]: Leaving directory '/dev/shm/heads/build/musl-cross-git'
make[1]: *** [Makefile:201: musl-cross.intermediate] Error 1
make[1]: Leaving directory '/dev/shm/heads'
Makefile:386: recipe for target 'all' failed
make: *** [all] Error 2
```

osresearch, 2017-03-31:
Drat.  I'm in the process of overhauling parts of the makefiles to make them configurable (issue #144) and probably broke it.

osresearch, 2017-03-31:
My clean checkout fails with a different error:

```
No rule to make target '/e/thudson/clean/heads/build/kexec-tools-2.0.12/build/sbin/kexec', needed by '/tmp/tmp.mGB8qMMyw6/bin/kexec'.  Stop.
```

So I've definitely broken something.

osresearch, 2017-03-31:
`kexec` was a different problem: https://github.com/osresearch/heads/commit/4141c75c8c42549fc7007eb9f10f7fadb5ea6478

Looking at your error, it appears to be in the `musl-cross` script, which I'm not super familiar with.  Are you building it in a ram disk?  Is there sufficient space?

paulmenzel, 2017-03-31:
Thank you for fixing the `kexec` problem. I hit that too.

Yes, I am building in RAM, and there is enough room with 128 GB.

```
$ ls -l crossgcc/x86_64-linux-musl/x86_64-linux-musl/
total 0
lrwxrwxrwx 1 joey joey 1 Mar 31 23:11 usr -> .
```

I can try to bisect it next week.

osresearch, 2017-04-06:
Is this still an issue? I've written a [building heads howto](http://osresearch.net/Building) and did multiple  builds as part of setting up a cloud builder on a brand new virtual machine, but didn't run into this problem.

paulmenzel, 2017-04-06:
The error is gone now. I updated to the latest commit 830828f (enable usb storage module (issue #160)), and noticed that `git clean -dfx`  doesn’t delete the git repository in `build/`.

```
$ git clean -dfx
Removing build/kernel-headers-3.12.6-5/
Removing build/log/
Removing build/make-4.2/
Skipping repository build/musl-cross-git/
Skipping repository build/tpmtotp-git/
Removing crossgcc/
Removing install/
Removing packages/.kernel-headers_verify
Removing packages/kernel-headers-3.12.6-5.tar.gz
Removing packages/make-4.2.tar.bz2
```

`rm -rf build/*` deleted those, and now the error is gone running `make BOARD=qemu MAKE_JOBS="-j8"`.

PS: Now I hit the problem described in issue #164.

paulmenzel, 2017-04-06:
Sorry, I am still hitting this. It was just masked by issue #164 when building in parallel.

I hit it again with `make BOARD=qemu MAKE_JOBS="-j1"`.

paulmenzel, 2017-08-31:
I just tried it again with v0.2.0-41-g9d9af31 (fix typo and format with markdown (issue #206)) and it works now.

The reason can be, that the musl files where actually not really cleaned. I think, I had to do `git clean -dffx`.

> Git will refuse to delete directories with .git sub directory or file unless a second -f is given.

closed 2017-08-31 by paulmenzel