linuxboot/heads #146: Document the PCRs

osresearch, 2017-03-31:
How does Heads use the PCRs?

geedo0, 2017-04-04:
I'd like to help out with this. Is there anything in particular that needs investigating? It's unclear to me whether this issue is about just writing documentation or if it's more digging into the source and reverse engineering Heads' components.

osresearch, 2017-04-04:
We control all the writes to the PCRs, so it is mostly documentation, although there are some things that need to be fixed in the code as well (such as #13 and #15 as well as #150).

osresearch, 2017-04-05:
Started on document: http://osresearch.net/Keys (or https://github.com/osresearch/heads-wiki/blob/master/Keys.md for edits)

closed 2017-04-12 by osresearch

tlaurion, 2019-04-29:
@geedo0 : if your help offer is still actual, #150 and #15 would need review!