linuxboot/heads #152: Expected hashes published somewhere?

jpouellet, 2017-04-01:
Reproducible builds are great, but only useful if you can compare against other peoples' results.

Is there anywhere where the expected results are published (and signed)?

I have obtained:
```
$ sha256sum *.rom
8f1c0f72e1517a22d48140398f75b5e676126fc25740eb63fb438c8155ef25ea  qemu.rom
c2bea0a4fe2d3ecc0504b4a7f49f40becc6ea8b541aabe7fd8777344c7733fb3  x230.full.rom
f1d96d9761cd2381160fb4c38d799f503634fcae22d076baf1741cda51dffd7e  x230.rom
```
from heads 93a0d7eee2642463a57d35b700c1a775f4e8911c

osresearch, 2017-04-02:
They are published for releases: https://github.com/osresearch/heads/releases/tag/v0.1.0

I haven't been publishing them each commit since I'm not doing clean builds and typically have working files in my `initrd` directories that would pollute the hashes.

closed 2017-04-02 by osresearch

jpouellet, 2017-04-02:
Great, thanks!