linuxboot/heads #157: SPI programmer recommendation

jpouellet, 2017-04-04:
I'm sure [your SPI programmer](https://trmm.net/SPI) works great, but for those who do not happen to have a teensy on hand and would end up buying some hardware to flash things anyway, I would recommend picking up a CH341A-based ([datasheet](http://www.datasheetspdf.com/datasheet/CH341A.html)) programmer. These:
- can read & write common SPI & I2C EEPROMs
- can be acquired for ~$3 USD including taxes and shipping (!) [on ebay](http://www.ebay.com/sch/i.html?_nkw=CH341A)
- have a [working open source driver](https://github.com/setarcos/ch341prog) to interface with them
- are decently fast: writes my 8mb ME region in ~100 seconds, reads it in ~60s, erases it in ~25s

SOIC8 clips can also be acquired (w/ tax & shipping!) for about $3 USD [on ebay](http://www.ebay.com/sch/i.html?_nkw=SOIC8+clip).

ciaby, 2017-04-05:
I second that, I used basically the same programmer. :)

osresearch, 2017-04-07:
That sound great.  Would you take a photo of it in operation and add details to the [Installing Heads](http://osresearch.net/Installing-Heads) page?

I also need to update those instructions for the new `flashrom` based install; there is a two-phase process of flashing the 4MB image to get a Heads bootstrap image that can then mount a USB device and flash a larger image.  This is also how subsequent upgrades can be performed without having to access the hardware again.

tlaurion, 2017-05-26:
Actually, it is supported through flashrom.
`./flashrom -p ch341a_spi`

jpouellet, 2017-06-06:
@osresearch I see you're linking to external images in the heads-wiki repo instead of hosting them there. Is there a reason to prefer that over having it self-contained?

Anyway, I don't have a fancy camera, but here are some pictures of a ch341a-based programmer in use:
[image]
[image]
[image]

osresearch, 2017-06-06:
I bought that same [SOIC clip on amazon](https://www.amazon.com/CTYRZCH-SOIC8-Socket-Adpter-Programmer/dp/B015W4PKR6) for $10 and it is nowhere near as nice as the $13 [Pomona 5250](https://www.digikey.com/product-detail/en/pomona-electronics/5250/501-1311-ND/745102).  Unfortunately even those seem to wear out after a few hundred cycles.

You're right that the images should be self contained. They were on flickr when it was hosted on trmm.net so that I didn't have to pay for the bandwidth...

jpouellet, 2017-06-06:
I got my clip for ~$3, but it can be quite frustrating to get it to make a good connection. I've just ordered the one you suggest off digikey, hopefully it's better. Clips in general are pretty annoying, and I almost considered soldering some magnet wire to the pins or something.

Images locally here: https://github.com/osresearch/heads-wiki/pull/2

mfc, 2024-01-15:
fyi libreboot folks have strong opinions on the ch341a: https://libreboot.org/docs/install/spi.html#do-not-use-ch341a just sharing so folks are aware since in heads documentation ch341a is the default flasher. (i re-purpose an old rpi 3 model B which works fine for me)

tlaurion, 2024-01-15:
> fyi libreboot folks have strong opinions on the ch341a: https://libreboot.org/docs/install/spi.html#do-not-use-ch341a just sharing so folks are aware since in heads documentation ch341a is the default flasher. (i re-purpose an old rpi 3 model B which works fine for me)

Yeah, had discussions with Leah about that. Main points of her were that assembly is cheap but when confronted about v1.6/1.7 board revision voltage selection, she stayed opinionated outside of the facts. Can you comment on https://github.com/linuxboot/heads-wiki/issues/120#issuecomment-1741039038 ? I plan on streamlining and taking new pictures to properly document this in issue prior of putting it in the wiki. 

Facts are exposed there, where rainbow cable pictures are still coupled with ch341a v1.2 old board revision, which I programmed hundreds of x230 without issues. But still risky with 5v on old < v1.5 boards where volt selector resolves the issue altogether.

zifxify, 2024-01-23:
I bought mine years ago after reading recommendations on winraid.
It looks just like this one (3.3v) and is good quality.
[image]

This links might be useful:
[https://winraid.level1techs.com/t/guide-using-ch341a-based-programmer-to-flash-spi-eeprom/30834/82](url)
[https://winraid.level1techs.com/t/guide-using-ch341a-based-programmer-to-flash-spi-eeprom/30834/205](url)