linuxboot/heads #164: Error building pciutils 3.5.4

paulmenzel, 2017-04-06:
With commit 830828f (enable usb storage module (issue #160)) the command below fails to build the PCI utilities 3.5.4.

```
$ make BOARD=qemu MAKE_JOBS="-j8"
```

Here is the excerpt from the log file.

```
$ more /dev/shm/heads/build/log/pciutils.log
[…]
make[3]: Entering directory '/dev/shm/heads/build/pciutils-3.5.4/lib'
/dev/shm/heads/install/bin/musl-gcc -fdebug-prefix-map=/dev/shm/heads=heads -gno-record-gcc-switches  -O2 -Wall -W -Wno-parentheses -Wstrict-prototypes -Wmissing-prototypes -fPIC -fvisibility=hidden   -c -o init.o init.c
ls-kernel.c:21:21: fatal error: libkmod.h: No such file or directory
compilation terminated.
make[2]: *** [<builtin>: ls-kernel.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[…]
```

The header file from kmod is there though.

```
/usr/include/libkmod.h
```

osresearch, 2017-04-06:
I think this is due to `BOARD=` no longer being sufficient to configuration the output. I wrote some documentation on the build process here: http://osresearch.net/Building

Try running without any `BOARD` or `CONFIG` on the make command line; it will default to `CONFIG=config/qemu-moc.config`, which should be sufficient to test.

osresearch, 2017-06-13:
It looks like the `pciutils/lib/configure` is a home made script that doesn't handle cross compilation. The ubuntu build host doesn't have a `libmkod.h`, so their configure is not setting `PCI_USE_LIBKMOD`, but the gentoo machine has it and that option gets set. Since nothing from `/usr/include` is in the path during compilation (we don't want to depend on the build machine headers), this is causing the problem.

Can you try this patch?

```
iff --git a/modules/pciutils b/modules/pciutils
index 2f0ab57..94bf65c 100644
--- a/modules/pciutils
+++ b/modules/pciutils
@@ -21,6 +21,7 @@ pciutils_target := \
        $(CROSS_TOOLS) \
        ZLIB=no \
        HWDB=no \
+       LIBKMOD=no \
        SHARED=yes \
        IDSDIR="/" \
        PREFIX="/" \
@@ -30,6 +31,7 @@ pciutils_target := \
        $(CROSS_TOOLS) \
        ZLIB=no \
        HWDB=no \
+       LIBKMOD=no \
        SHARED=yes \
        PREFIX="/" \
        DESTDIR="$(INSTALL)" \
```

rotation3, 2017-06-13:
Hi,

Great! This works after running 'make pciutils.clean'. Just editing the module/pciutils file or applying the patch to results in the same problem if an existing build is not cleaned of cached objects.

I notice that the build system doesn't reconfigure a module each time it rebuilds them. This wouldn't take long and wouldn't cause a problem for a module that has been build already, if the module just checks for cached objects and doesn't rebuild them if there is no configuration change.

Thanks!

osresearch, 2017-06-13:
For submodules that use real `configure` scripts, they are re-configured if the Heads `config/foo.config` file is updated.  Unfortunately pciutils is not one of those; it runs its "configure" script as part of its makefile, so we don't have a distinct configuration phase.

closed 2017-06-13 by osresearch