linuxboot/heads #176: Error building kexec

geedo0, 2017-04-09:
Checked out at commit 132d26de05590e94ea6539e4d3cbf621c721005a

Using the command 'make CONFIG=config/x230-moc.config' the build fails making the target build/sbin/kexec.

```
tail /foo/heads/build/log/kexec.log
-----
heads/build/kexec-tools-2.0.12/kexec/arch/i386/kexec-elf-x86.c:192: undefined reference to `purgatory'
kexec/arch/i386/kexec-bzImage.o: In function `do_bzImage_load':
heads/build/kexec-tools-2.0.12/kexec/arch/i386/kexec-bzImage.c:206: undefined reference to `purgatory_size'
heads/build/kexec-tools-2.0.12/kexec/arch/i386/kexec-bzImage.c:206: undefined reference to `purgatory'
heads/build/kexec-tools-2.0.12/kexec/arch/i386/kexec-bzImage.c:209: undefined reference to `purgatory_size'
heads/build/kexec-tools-2.0.12/kexec/arch/i386/kexec-bzImage.c:209: undefined reference to `purgatory'
kexec/arch/i386/kexec-multiboot-x86.o: In function `multiboot_x86_load':
heads/build/kexec-tools-2.0.12/kexec/arch/i386/kexec-multiboot-x86.c:218: undefined reference to `purgatory_size'
heads/build/kexec-tools-2.0.12/kexec/arch/i386/kexec-multiboot-x86.c:218: undefined reference to `purgatory'
kexec/arch/x86_64/kexec-elf-x86_64.o: In function `elf_x86_64_load':
heads/build/kexec-tools-2.0.12/kexec/arch/x86_64/kexec-elf-x86_64.c:188: undefined reference to `purgatory_size'
heads/build/kexec-tools-2.0.12/kexec/arch/x86_64/kexec-elf-x86_64.c:188: undefined reference to `purgatory'
heads/build/kexec-tools-2.0.12/kexec/arch/x86_64/kexec-elf-x86_64.c:188: undefined reference to `purgatory_size'
heads/build/kexec-tools-2.0.12/kexec/arch/x86_64/kexec-elf-x86_64.c:188: undefined reference to `purgatory'
kexec/arch/x86_64/kexec-bzImage64.o: In function `do_bzImage64_load':
heads/build/kexec-tools-2.0.12/kexec/arch/x86_64/kexec-bzImage64.c:174: undefined reference to `purgatory_size'
heads/build/kexec-tools-2.0.12/kexec/arch/x86_64/kexec-bzImage64.c:174: undefined reference to `purgatory'
collect2: error: ld returned 1 exit status
make[2]: *** [kexec/Makefile:100: build/sbin/kexec] Error 1
make[2]: Leaving directory '/foo/heads/build/kexec-tools-2.0.12'
make[1]: *** [Makefile:231: kexec.intermediate] Error 1
make[1]: Leaving directory '/foo/heads'
Makefile:452: recipe for target 'all' failed
make: *** [all] Error 2
```

osresearch, 2017-04-09:
Try (in increasing rebuild times):

* `make kexec.clean`
* `rm -rf build/kexec-tools-2.0.12`
* `make real.clean`

I noticed that kexec and a few other things were using the `/usr/bin/ar` instead of our cross compiler, so if you have any leftover libraries they might have been archived with the wrong tool.

geedo0, 2017-04-09:
`make real.clean` fixed my problems. Thanks!

closed 2017-04-09 by geedo0