linuxboot/heads #177: coreboot-4.5/crossgcc build fails w/ gcc 6.2.0 (Ubuntu 16.10 default)

flammit, 2017-04-09:
The config.log error is below:

```
configure:5757: checking for the correct version of the gmp/mpfr/mpc libraries
configure:5788: cc -o conftest -O2  -fomit-frame-pointer -m64 -I/home/flam/devel/thinkpad/heads/build/coreboot-4.5/util/crossgcc/xgcc/include -I/home/flam/devel/thinkpad/heads/build/coreboot-4.5/util/crossgcc/xgcc/include -I/home/flam/devel/thinkpad/heads/build/coreboot-4.5/util/crossgcc/xgcc/include    conftest.c  -L/home/flam/devel/thinkpad/heads/build/coreboot-4.5/util/crossgcc/xgcc/lib -L/home/flam/devel/thinkpad/heads/build/coreboot-4.5/util/crossgcc/xgcc/lib -L/home/flam/devel/thinkpad/heads/build/coreboot-4.5/util/crossgcc/xgcc/lib -lmpc -lmpfr -lgmp >&5
/usr/bin/ld: /home/flam/devel/thinkpad/heads/build/coreboot-4.5/util/crossgcc/xgcc/lib/libgmp.a(fat_entry.o): relocation R_X86_64_32S against symbol `__gmpn_cpuvec' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: Nonrepresentable section on output

...

configure:5792: result: no
configure:5813: error: Building GCC requires GMP 4.2+, MPFR 2.4.0+ and MPC 0.8.0+.
Try the --with-gmp, --with-mpfr and/or --with-mpc options to specify
their locations.  Source code for these libraries can be found at
their respective hosting sites as well as at
ftp://gcc.gnu.org/pub/gcc/infrastructure/.  See also
http://gcc.gnu.org/install/prerequisites.html for additional info.  If
you obtained GMP, MPFR and/or MPC from a vendor distribution package,
make sure that you have installed both the libraries and the header
files.  They may be located in separate packages.
```

This stems from a resolved issue in coreboot: https://github.com/coreboot/coreboot/commit/8bbd596de631adc8b677e69603e978b848eb1708

I've locally tested an updated coreboot-4.5.patch successfully and will submit a PR in a bit.

osresearch, 2017-04-09:
I'm testing your patch on 16.10 now -- I had noticed that it wouldn't build on a digitalocean 16.10 droplet as part of tracking down some reproducibility issues in #166, but tabled it for another day.

osresearch, 2017-04-09:
It's unfortunate that the build process takes so long on a clean checkout... I ran on both fedora and ubuntu VM builds and have both successful gcc binaries as well as ROM images with matching hashes.

closed 2017-04-09 by osresearch

flammit, 2017-04-10:
Thanks for the quick merge and all the great work on this project.