linuxboot/heads #180: supporting of Lenovo x220 laptop

DarkCoridor, 2017-04-10:
I understand that x230 is only currently supported. I am wondering about the difficulty of me making Heads work with the x220 Lenovo also. Looking at the repo, there are the various config files like /config/x230-flash.config which I assume modifying these is part of the process.

Any documentation tips or suggestions on things to do so I can make x220 supported would be very helpful. Thank you.

osresearch, 2017-04-10:
There is a brief guide on how to add support for a new board: http://osresearch.net/Porting

If you have a working coreboot `.config` file it shouldn't be too difficult.  Do you know how much space is free in the x220 spi flash? It looks like it is 8 MB, which should be sufficient depending on how much is taken up by the ME.

osresearch, 2017-04-10:
The `x230-flash.config` is a helper to bootstrap installation on the two-SPI chip x230.  Since it is a pain to do an initial flash of both chips (you have to disconnect and reconnect the programmer), I have a 4MB image that fits in the top chip and has sufficient functionality to mount a USB drive and run flashrom from it that will install the full featured x230 build.

DarkCoridor, 2017-04-13:
Good helpful information. I do have a "working coreboot `.config` file" that was extracted from my x220 BIOS so I will proceed that way and will report back my progress. Thank you @osresearch

jgrip, 2017-04-17:
The x220 does indeed use an 8MB SPI flash. The ME image from factory takes up ~5.3MB of that. Neutered ME image can be trunkated down to ~86kB so there should be plenty of space left.
I was looking at this earlier but I can't figure out a way in the build system to include the necessary ME, GBE and flash layout files during the build.

osresearch, 2017-04-17:
For the Chell chromebook and Librem 13, the coreboot `CONFIG_ME_BIN_PATH` and `CONFIG_IFD_BIN_PATH` values point to the correct blobs. That's not ideal, but should be workable to build a flashable image.

If you have a system that is already booting and can run `flashrom`, you can use a layout file to allow a partial flash of just the BIOS region.  This is what [initrd/bin/flashrom-x230.sh](https://github.com/osresearch/heads/blob/master/initrd/bin/flashrom-x230.sh) does, since the `x230.rom` image has no IFD, ME or GBE sectoins.

jgrip, 2017-04-18:
I created the config files and used my existing coreboot config with some tweaks. Seems to work as it should, trying to install qubes now. For now I just hardwired the blobs to a folder in my home directory. I guess we can't easily distribute these as they are not open source.

jgrip, 2017-04-21:
See #190

osresearch, 2017-05-01:
#190 has been merged. I'm not able to test it (or even build it), so hopefully it works.

closed 2017-05-01 by osresearch