linuxboot/heads #185: What is the best place to discuss potential new development/features?

flammit, 2017-04-13:
Should we propose and discuss work as issues here on Github or is there another mechanism more appropriate (IRC/gitter/slack)?

osresearch, 2017-04-17:
Github issues have been were most of the discussion has occured thus far; it has a decent searchable log and history, and is public.  Slack has the problem of messages disappearing after 10k posts and not being public. irc works well for some people, although the lack of history is problematic and I've found that I don't hang out enough anymore. I haven't tried gitter; do you have any experience with it?

flammit, 2017-04-17:
Sorry I don't have any experience with gitter and I agree on the IRC and Slack points.  Using Github issues seems like a perfectly fine option, esp for dev issues, but I just wanted to confirm that I didn't miss a contrib guideline or user group somewhere.  

Feel free to close this out unless you wanted to use the issue to update the wiki or README.md later.

closed 2017-06-24 by flammit