linuxboot/heads #186: EHCI kernel modules compiled for the wrong kernel

geedo0, 2017-04-15:
I checked out and built the 0.2.0 release of Heads using the commands below and I can't get the USB kernel modules to load anymore.
```
git checkout -f v0.2.0
make modules.clean
make real.clean
make CONFIG=config/x230-flash.config
```
After flashing the new firmware onto an X230 via the SPI flasher and booting into Heads I get the following:
```
insmod /lib/modules/ehci-hcd.ko
ehci_hcd: version magic '4.9.7-heads_moc SMP ' should be '4.9.20-heads_moc SMP '
insmod: can't insert '/lib/modules/ehci-hcd.ko': invalid module format
```
Let me know if I'm missing something obvious or if there's anything I can try. Thanks!

osresearch, 2017-04-15:
That is... surprising.  The 0.2.0 initrd shouldn't end up with any files that were previously built, so my guess is that you had a pre-release build that had copied the old modules into `heads/initrd/lib/modules/` and those overwrote the newly built ones.

If you preserve any changes you have in `heads/initrd` and then `rm -rf initrd; git checkout initrd` I would wager it will work.

geedo0, 2017-04-16:
That fixed the problem for me, but does this mean there's a bug in the clean targets? It might be best to just do a git clean -dxf and start fresh if anyone has stuff built prior to the v0.2.0 Makefile changes.

One interesting thing I noticed is that I get this every time I run a build:
`2017-04-16 08:15:54-04:00 Wrong make detected: 4.1`

osresearch, 2017-04-17:
There were some major changes in the lead up to 0.2.0 that definitely required a clean checkout or a manual pruning of the file system to build correctly.  We might have a patch coming soon (related to #188 ) since we're moving to use the headers from our build kernel rather than the 3.x headers that we had been unpacking into `install/include`.

This one should be "fixed" by a `make real.clean`, without requiring a full removal of the `build/` subdirectory, although that should always be a safe, if time consuming, step.

closed 2017-04-17 by osresearch

osresearch, 2017-04-19:
Following up on the `wrong make detected` warning, one of the first submodules built is `make 4.2` to ensure that we have reproducible build rules.  On older Ubuntu systems (12?) the make 3 didn't parse the makefile correctly, leading to broken builds.  So if the top level make detects that it is not running our version of make, it re-invokes the makefile with that one.

We can probably silence the warning, since I put it in there for debugging while writing that part of the system.