linuxboot/heads #187: Include ZFS

jonathanfemideer, 2017-04-15:
At least [one](https://github.com/osresearch/heads/issues/77) other person is interested in using Heads with [ZFS](https://en.wikipedia.org/wiki/OpenZFS).

If there is not enough room in the X230 ROM for both ZFS *and* LVM functionality, then perhaps the user could be empowered to choose one or the other, as they are unlikely to need both.

osresearch, 2017-04-15:
I looked into building zfs (and spl) and it looks like there is a complication with building modules.  `configure` fails on that step with this error:

```
configure:15600: cp conftest.c build && make modules -C ../linux-4.9.20 EXTRA_CFLAGS=-Werror-implicit-function-declaration   M=/build/heads/build/spl-0.6.5.9/build 
make[3]: *** No rule to make target 'tools/objtool/objtool', needed by '/build/heads/build/spl-0.6.5.9/build/conftest.o'.  Stop.
make[2]: *** [_module_/build/heads/build/spl-0.6.5.9/build] Error 2
```

Since the kernel was cross compiled, we don't have the host tools built.

osresearch, 2017-04-15:
With the `objtool` built there are lots of errors in `spl/module/splat/splat-cred.c` that prevent it from building.  I haven't debugged it further.

```
/build/heads/build/spl-0.6.5.9/module/spl/spl-cred.c: In function 'crgetngroups':
/build/heads/build/spl-0.6.5.9/module/spl/spl-cred.c:94:11: error: 'NGROUPS_PER_BLOCK' undeclared (first use in this function)
  if (rc > NGROUPS_PER_BLOCK) {
           ^
/build/heads/build/spl-0.6.5.9/module/spl/spl-cred.c:94:11: note: each undeclared identifier is reported only once for each function it appears in
/build/heads/build/spl-0.6.5.9/module/spl/spl-cred.c: In function 'crgetgroups':
/build/heads/build/spl-0.6.5.9/module/spl/spl-cred.c:119:8: error: 'struct group_info' has no member named 'nblocks'
  if (gi->nblocks > 0)
        ^
In file included from /build/heads/build/spl-0.6.5.9/module/spl/spl-cred.c:27:0:
/build/heads/build/spl-0.6.5.9/module/spl/spl-cred.c:120:27: error: 'struct group_info' has no member named 'blocks'
   gids = KGIDP_TO_SGIDP(gi->blocks[0]);
                           ^
/build/heads/build/spl-0.6.5.9/include/sys/cred.h:60:30: note: in definition of macro 'KGIDP_TO_SGIDP'
 #define KGIDP_TO_SGIDP(x) (&(x)->val)
                              ^
make[7]: *** [scripts/Makefile.build:293: /build/heads/build/spl-0.6.5.9/module/spl/spl-cred.o] Error 1
```

osresearch, 2017-04-15:
I've pushed the submodules files to the `zfs` branch if you want to try to make it work.